mantidproject/mantid #33494: Redhat Plotting Smoke Tests

DanielMurphy22, 2022-02-08:
## If at ISIS, please test on IDAaaS
* Follow the instructions [here](https://stfc365.sharepoint.com/:w:/s/isiscomputingdivision-staffsite/ESTnjvxoislDsmpI8vxNyLUBZxw_51y-Qw0OklU_vNgXVA?e=Ih3iLK) to test Mantid on IDAaaS

Before testing:
- Check this testing issue relates to the OS you will test on.
- If unassigned, please assign yourself as for a normal Github issue.
- Please run these tests on the release package of Mantid; **not a locally built version**.

Afterwards:
- Comment below with any issues you came across.
- If no issues were found, or they are now all resolved, please close the testing issue.
- Check the master issue for this OS for other unassigned smoke tests.

If you have any questions please contact the creator of this issue.

 :chart_with_upwards_trend: :bar_chart:

* 1D plotting:[instructions](http://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/loading_and_displaying_data/03_displaying_1D_data.html#displaying-1d-data)
 - [ ] Simple plot 
 - [ ] Another way to plot 
 - [ ] Adding curves to existing plots 
 - [ ] Also, test out [waterfall](https://docs.mantidproject.org/nightly/plotting/WaterfallPlotsHelp.html#waterfall-plots) and [tiled]( https://docs.mantidproject.org/nightly/plotting/1DPlotsHelp.html#tiled-plots)
 - [ ] Check Toolbar buttons

* 2D plotting: [instructions](http://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/loading_and_displaying_data/04_displaying_2D_data.html#displaying-2d-data)
 - [ ] Plot all spectra 
 - [ ] Change colour map 
 - [ ] [Contour plot](https://docs.mantidproject.org/nightly/plotting/ColorfillPlotsHelp.html#contour-plots) (under 3D menu) 
 - [ ] Check Toolbar buttons

* [3D plotting](https://docs.mantidproject.org/nightly/plotting/3DPlotsHelp.html):
 - Load some data eg `LOQ7041` from the ISIS sample data
 - [ ] 3D surface
 - [ ] 3D wire frame  
 - [ ] Check Toolbar buttons

 ## Sliceviewer
 - [ ] Overly long instructions (don't spend  long!) and data [here](https://developer.mantidproject.org/Testing/SliceViewer/SliceViewer.html). In particular try editing the data in a workspace while it is open in Sliceviewer!

sf1919, 2022-02-08:
Can we update the plotting documentation? It's not clear how to access waterfall or tiled plots.

sf1919, 2022-02-08:
On the Contour plot the copy script button doesn't work.

thomashampson, 2022-02-08:
> On the Contour plot the copy script button doesn't work. I'm also have the same problem with 3D surface plots and wireframe.

I think the script button should be hidden from the toolbar for 3D plots, so if you're seeing it then that's a bug.

sf1919, 2022-02-08:
The scripting button isn't there - I was trying to use the copy image to clipboard button which probably does something but it's not obvious!

closed 2022-02-08 by sf1919