mantidproject/mantid #33495: Redhat Python Smoke Tests

DanielMurphy22, 2022-02-08:
## If at ISIS, please test on IDAaaS
* Follow the instructions [here](https://stfc365.sharepoint.com/:w:/s/isiscomputingdivision-staffsite/ESTnjvxoislDsmpI8vxNyLUBZxw_51y-Qw0OklU_vNgXVA?e=Ih3iLK) to test Mantid on IDAaaS

Before testing:
- Check this testing issue relates to the OS you will test on.
- If unassigned, please assign yourself as for a normal Github issue.
- Please run these tests on the release package of Mantid; **not a locally built version**.

Afterwards:
- Comment below with any issues you came across.
- If no issues were found, or they are now all resolved, please close the testing issue.
- Check the master issue for this OS for other unassigned smoke tests.

If you have any questions please contact the creator of this issue.

 :snake:

* Test that the Python scripting window works, [directions here](https://docs.mantidproject.org/nightly/workbench/scriptwindow.html)
 - [ ] Editor options 
 - [ ] Execution options 
 - [ ] Script output  
- [ ] Perform some workspace algebra 
- [ ] Test numpy functionality 
- [ ] Use the scripting window to run some scripts 
- [ ] Run through some examples from [the documentation](https://docs.mantidproject.org/nightly/tutorials/python_in_mantid/index.html), 3 or 4 examples from the Solutions is enough

RichardWaiteSTFC, 2022-02-08:
Some keyboard shortcuts not working on IDAaaS: `ctrl-n` (opens new tab in chrome) and `ctrl-shift-return ` (doesn't abort running script)

RichardWaiteSTFC, 2022-02-08:
Have found nothing release breaking

closed 2022-02-08 by RichardWaiteSTFC