mantidproject/mantid #33498: Ubuntu Fitting Smoke Tests

DanielMurphy22, 2022-02-08:
* Follow the instructions [here](https://download.mantidproject.org/ubuntu.html) to install the .deb for Ubuntu

Before testing:
- Check this testing issue relates to the OS you will test on.
- If unassigned, please assign yourself as for a normal Github issue.
- Please run these tests on the release package of Mantid; **not a locally built version**.

Afterwards:
- Comment below with any issues you came across.
- If no issues were found, or they are now all resolved, please close the testing issue.
- Check the master issue for this OS for other unassigned smoke tests.

If you have any questions please contact the creator of this issue.

 :muscle: :runner:

* Fitting to data [directions](http://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/fitting_data/01_fitting_models_to_data.html#fitting-models-to-data)
- [ ] Simple fit 
- [ ] Activating the fitting tools 
- [ ] Selecting the fitting range 
- [ ] Other Fit Function Settings properties 
- [ ] Setting up a fit model 
- [ ] Adjusting fit function parameter 
- [ ] Saving a model 
- [ ] Tying and constraining fit parameters 
- [ ] Execute your fit 
- [ ] Inspecting a fit result 
* Additional options [directions](https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/fitting_data/03_fit_model_choices.html)
- [ ] Non peak model + background 
- [ ] User defined function

Pasarus, 2022-02-08:
Whilst fitting you can achieve a light crash where the error reporter opens on you (OS agnostic):
- Open a workspace plot
- Open the fit menu
- add a flat background
- Click display -> plot guess
- Then add a new function, add User function (it should fall over)

Pasarus, 2022-02-08:
Issues raised as #33505

closed 2022-02-08 by Pasarus