mantidproject/mantid #33505: User function crash

AnthonyLim23, 2022-02-08:
If you add a user function then press fit or show guess, an error will appear in the logger. The next time you press either button Mantid crashes

Open Mantid
Load some data
Plot the data
Press the "fit" button to get the the fit browser to appear
Add a user function
Press "fit"
An error will appear in the logger
Press "fit" again - crash

thomashampson, 2022-08-17:
I've updated the description because I couldn't reproduce the crash without some key information.

thomashampson, 2023-07-27:
We had some people come to the office to report this too, so it's probably worth fixing.

sf1919, 2023-07-28:
This also occurs when undertaking the following steps on v6.7

1. Open Mantid
2.  Open Muon Analysis GUI
3. Select EMU as instrument (if not already the default)
4. Load `EMU00020884.nxs` from the Training Course Data set
5. Click on the fitting tab
6. Add a Fitting function (e.g, Dynamic Kubo Toyabe)
7. Tick Plot guess
8. Untick plot guess
9. Add another fitting function
10. Click plot guess

Mantid then crashes.

AnthonyLim23, 2023-07-31:
It looks like the original issue has been fixed (since it was in the main fitting of Mantid, i.e. not in the muon GUI). 

The new issue seems to only happen in the muon GUI. I have been unable to replicate it in
1. The main fitting window
2. Indirect data analysis
3. Enginex
4. ALC

so here is probably a good place to start looking 
https://github.com/mantidproject/mantid/blob/main/qt/python/mantidqtinterfaces/mantidqtinterfaces/Muon/GUI/Common/fitting_widgets/basic_fitting/basic_fitting_presenter.py#L456

PR #35972 (2023-09-04): Fix crash in muon analysis fitting interface
The bug described in issue #33505 is an exception thrown in the destructor of either CreateWorkspace or EvaluateFunction. EvaluateFunction was being called directly in the muon analysis fitting GUI without going through the AlgorithmManager, which I think is the source of the problem. So I used the AlgorithmManager instead.

**Description of work**
- Used AlgorithmManager for calling EvaluateFunction
- Moved creation of workspace into smaller scope

**Purpose of work**

Fixes #33505 

**Summary of work**

**To test:**
Original bug in #33505, instruc [... cut]
closed 2023-09-04 by SilkeSchomann