mantidproject/mantid #33510: Update cppcheck to v2.6

thomashampson, 2022-02-08:
Cppcheck should be updated so we can benefit from improvements. Needs to be updated on the Jenkins builder and in the conda developer .yml files. Also need to update suppressions list if required.

thomashampson, 2022-03-08:
2.6.2, 2.6.3 and 2.7 all have internal cppcheck errors, e.g.:
```
        </error>
        <error id="cppcheckError" severity="error" msg="Internal error: Child process crashed with signal 11" verbose="Internal error: Child process crashed with signal 11">
            <location file="/mantid_src/Framework/DataObjects/src/MDBoxSaveable.cpp " line="0" column="0"/>
        </error>

```

thomashampson, 2022-08-18:
2.7.5 fails to work:
`#### ThreadExecutor::handleRead error, type was:i`

Also, oddly it prints out `-- cppcheck version: 2.7.4` near the start.

There are still internal cppcheck errors in thisd version too.

sf1919, 2023-02-16:
cppcheck is now on v2.10. We need to be on v2.8 or higher to upgrade `pygments` (e.g. https://github.com/mantidproject/mantid/blob/main/mantid-developer-win.yml#L52). We need to update `pygments` in order to upgrade Sphinx (#34584)

thomashampson, 2023-03-21:
We can try v2.10 now that it's on conda-forge

thomashampson, 2023-07-13:
Suggestion from Martyn:
For the cppcheck issue with the test files one option might be to add a custom target that just generates the test files and doesn't compile them. Using add_custom_target (https://cmake.org/cmake/help/latest/command/add_custom_target.html)  with DEPENDS and the list of cpp files in https://github.com/mantidproject/mantid/blob/main/buildconfig/CMake/FindCxxTest.cmake#L130 would mean you could make the cppcheck cmake target depend on these targets so it will generate the sources first and then run cppcheck without having to compile the tests too.

PR #35810 (2023-07-19): update cppcheck to 2.10.3
**Description of work**
Here we update cppcheck to the latest version available on conda-forge. The following list details the steps required to make it work:
1. The latest cppcheck requires all source files in the input `cmake_compile_commands.json` map to exist rather than simply issuing a warning. Here we create a separate target to run just the source cxxtest generation and and attach this as dependencies to the cppcheck target.
2. The `compile_commands.json` file contains various qt source files (e.g. `qrc_*.cpp` & `moc_\*.cpp`) that are generated by build rules  [... cut]
closed 2023-07-19 by SilkeSchomann