mantidproject/mantid #33543: We'll have to push this off to a separate issue, I agree this is a good idea but I'm worried about scope creep

smk78, 2022-02-10:
We'll have to push this off to a separate issue, I agree this is a good idea but I'm worried about scope creep
This PR has flagged up a number of things as we go through the docs with a fine-toothed comb

_Originally posted by @DavidFair in https://github.com/mantidproject/mantid/pull/33065#discussion_r800710754_

smk78, 2024-03-22:
Following a query from the ESS SANS Team and subsequent investigation by @smk78  it has been established that:

- there is no way to switch this correction on/off from TOML User Files at present
- but the necessary algorithm already exists in Mantid (https://docs.mantidproject.org/nightly/algorithms/SANSWideAngleCorrection-v1.html#index-0)
- although it does, however, generate an error in certain situations, possibly when the Sample/Can transmission and the Direct Beam transmission are one and the same
```
Computing SANSWideAngleCorrection
Invalid value for XMax: XMax must be greater than XMin
Invalid value for XMin: XMin must be less than XMax
Error in execution of algorithm CropWorkspace:
Some invalid Properties found: 
 XMax: XMax must be greater than XMin
 XMin: XMin must be less than XMax
Error in execution of algorithm SANSFitShiftScale:
Some invalid Properties found: 
 XMax: XMax must be greater than XMin
 XMin: XMin must be less than XMax
  at line 81 in 'C:/MantidInstall-6.9.0/plugins/python\algorithms/WorkflowAlgorithms/SANSFitShiftScale.py'
  caused by line 152 in 'C:/MantidInstall-6.9.0/plugins/python\algorithms/WorkflowAlgorithms/SANSFitShiftScale.py'
  caused by line 260 in 'C:/MantidInstall-6.9.0/plugins/python\algorithms/WorkflowAlgorithms/SANSFitShiftScale.py'
  caused by line 288 in 'C:/MantidInstall-6.9.0/plugins/python\algorithms/WorkflowAlgorithms/SANSFitShiftScale.py'
Error in execution of algorithm SANSStitch:
Some invalid Properties found: 
 XMax: XMax must be greater than XMin
 XMin: XMin must be less than XMax
  at line 81 in 'C:/MantidInstall-6.9.0/plugins/python\algorithms/WorkflowAlgorithms/SANSFitShiftScale.py'
  caused by line 152 in 'C:/MantidInstall-6.9.0/plugins/python\algorithms/WorkflowAlgorithms/SANSFitShiftScale.py'
  caused by line 260 in 'C:/MantidInstall-6.9.0/plugins/python\algorithms/WorkflowAlgorithms/SANSFitShiftScale.py'
  caused by line 288 in 'C:/MantidInstall-6.9.0/plugins/python\algorithms/WorkflowAlgorithms/SANSFitShiftScale.py'
  at line 344 in 'C:/MantidInstall-6.9.0/plugins/python\algorithms/WorkflowAlgorithms/SANSStitch.py'
  caused by line 240 in 'C:/MantidInstall-6.9.0/plugins/python\algorithms/WorkflowAlgorithms/SANSStitch.py'
Error in execution of algorithm SANSSingleReduction:
Some invalid Properties found: 
 XMax: XMax must be greater than XMin
 XMin: XMin must be less than XMax
  at line 81 in 'C:/MantidInstall-6.9.0/plugins/python\algorithms/WorkflowAlgorithms/SANSFitShiftScale.py'
  caused by line 152 in 'C:/MantidInstall-6.9.0/plugins/python\algorithms/WorkflowAlgorithms/SANSFitShiftScale.py'
  caused by line 260 in 'C:/MantidInstall-6.9.0/plugins/python\algorithms/WorkflowAlgorithms/SANSFitShiftScale.py'
  caused by line 288 in 'C:/MantidInstall-6.9.0/plugins/python\algorithms/WorkflowAlgorithms/SANSFitShiftScale.py'
  at line 344 in 'C:/MantidInstall-6.9.0/plugins/python\algorithms/WorkflowAlgorithms/SANSStitch.py'
  caused by line 240 in 'C:/MantidInstall-6.9.0/plugins/python\algorithms/WorkflowAlgorithms/SANSStitch.py'
  at line 131 in 'C:/MantidInstall-6.9.0/plugins/python\algorithms/WorkflowAlgorithms/SANS/SANSSingleReduction.py'
  caused by line 155 in 'C:\MantidInstall-6.9.0/plugins/python\algorithms/WorkflowAlgorithms/SANS\SANSSingleReductionBase.py'
  caused by line 303 in 'C:\MantidInstall-6.9.0\scripts\SANS\sans\algorithm_detail\single_execution.py'
  caused by line 131 in 'C:\MantidInstall-6.9.0\scripts\SANS\sans\algorithm_detail\merge_reductions.py'
```
- preliminary indications are that the correction is at the <=3% level

[image]

Enabling the algorithm by default, as illustrated above, carries the risk of reductions failing during edge case reductions, but may also lead to unacceptable reductions in workflow speed, so implementing a TOML switch is seen as the preferred option at this time. 

More information and test data is available on the ISIS SANS Sharepoint.

rbauststfc, 2025-01-16:
In recent prioritisation discussions this was described as "implementing a switch for WideAngleCorrection in the TOML file". There is apparently a GUI setting for applying the correction already but it just can't be controlled via the TOML file.