mantidproject/mantid #33551: Offer plotting in QSpace in Indirect Diffraction

sf1919, 2022-02-11:
**Is your feature request related to a problem? Please describe.**
When using Indirect Diffraction GUI there is the option to plot a spectrum after successfully loading a run on diffspec. At the moment the plot produced is in d-space. Users would like the option to have the results in Q-Space.

**Describe the solution you'd like**
Provide users with the ability to choose whether their plot is in d-space or Q-Space and produce the relevant plot.

**Additional context**
Requested by Users

sf1919, 2022-08-15:
@AnthonyLim23 I'll leave this one for you to decide on.

Harrietbrown, 2023-01-25:
[image] [Offer plotting in QSpace in Indirect Diffraction](https://trello.com/c/gdtWpgi3/20-offer-plotting-in-qspace-in-indirect-diffraction)

PR #35456 (2023-05-16): Add unit option to Indirection Diffraction UI output plot options
**Description of work.**
This PR adds an option to select the unit of the output plot in the indirect diffraction interface. Users can now choose from "D-spacing" or "Q-spacing" unit options. 

**To test:**
1- Open Indirect Diffraction interface (Interfaces->Indirect->Diffraction).
2- Change the Reflection setting to diffspec.
3- Click the Run button. 
4- Once the run is complete, verify that the unit combo box and the `Plot Spectra` button are both enabled.
5- Select the two possible units in the unit combo box and plot the spe [... cut]
closed 2023-05-16 by gemmaguest