mantidproject/mantid #33616: Investigate feasibility of modifying step size in fitting

SilkeSchomann, 2022-03-07:
**Describe the outcome that is desired.**
One of the outcomes of the ALC global optimization project was that the currently used step size in Mantid fitting is not suitable. It might be possible to improve the performance by modifying the step size as follows:

```
For the ith parameter x0 use
step = sqrt(epsilon) when either abs(x0_i) < 1 or x0_i = 0
step = sqrt(epsilon)*x0_i otherweise
```

Since fitting is used in various interfaces and also a large number of automated tests, the aim of this issue is to investigate the feasibility of this modification (as opposed to implementing it).

AnthonyLim23, 2022-03-08:
how do we currently calculate the step size?

SilkeSchomann, 2022-03-08:
`  constexpr double epsilon = std::numeric_limits<double>::epsilon() * 100;
  constexpr double stepPercentage = 0.001;
  constexpr double cutoff = 100.0 * std::numeric_limits<double>::min() / stepPercentage;`

and then for val as the ith active parameter

`      if (fabs(val) < cutoff) {
        step = epsilon;
      } else {
        step = val * stepPercentage;
      }`

AnthonyLim23, 2022-03-08:
I can only see it breaking in 2 circumstances:

1. the new steps are smaller and you hit the maximum number of iterations (at this point I think its probably safe to say that the start guess is rubbish)
2. the new steps are bigger so it misses a narrow minima. However, I suspect that would use the sqrt(epsilon) which would be a smaller step -> so would still work as long as point 1 does not happen.

robertapplin, 2022-04-12:
Discussed this briefly with @SilkeSchomann, and we think it might be a good idea, as a first step, to create a system test to benchmark the accuracy of different fitting problems in Mantid. To do this we could potentially use FitBenchmarking which can give us a benchmark for accuracy and runtime.

Then, when we modify the step size, this system test will tell us if there is an improvement or not

closed 2022-05-06 by DannyHindson