mantidproject/mantid #33638: Investigate polarised SANS workflow

gemmaguest, 2022-03-08:
@DavidFair to meet with Dirk to do some initial groundwork into polarised SANS requirements

smk78, 2022-03-09:
Dirk (@Dehoni) comments via email (09/03/2021 to @DavidFair, @smk78, & Diego):

As a discussion starter for a project to implement polarised SANS in Mantid, here is also a document   https://stfc365-my.sharepoint.com/:w:/g/personal/dirk_honecker_stfc_ac_uk/Eb5G595cOwVKqvJJByUpL2EBu-PUyZhuMXt2rf5YTiXs9Q
With a further description. It should summarize the requirements and steps that a polarisation correction for SANS might involve. It covers different polarised mode like SANSPOL (only polarised incoming beam no analysis after sample), full longitudinal polarisation analysis (PA) with a 3He gas analyser and a simplified PA for certain sample systems. The document might be overly detailed and based on first principles potentially reproducing what already exists in Mantid. There is certainly comparable work on other techniques, to simplify the task and not reinvent the wheel.

smk78, 2022-03-09:
TO NOTE: The above document does not explicitly consider SESANS. Since this is also a polarised SANS technique, it might be prudent to include this in any project scoping.

smk78, 2022-03-14:
Where this discussion leads us is towards a (near) future where there will be _at least_ 3 SANS reduction workflows:

- SANS; this is what already exists in Mantid
- SESANS
- Pol-SANS

SESANS is currently reduced using scripts devised by @rprospero , but it is desirable for Users to be able to use an interface as they do for SANS. The functionality of this interface is not so different to that of the current ISIS SANS interface, as in, there are still sample and can runs and there are still transmission runs. What changes is how you process those runs. The existing scripts, however, may need unpicking to incorporate them into the Mantid codebase.

Pol-SANS will require a greater degree of functionality conferred on the existing SANS interface to support the more involved reduction workflow outlined in the document above.

Talking to @rmdalgliesh there would seem to be a preference to have a single SANS interface - rather than a separate SANS interface, a SESANS interface, and a Pol-SANS interface - but with functionality included/removed as befits the type of measurement. Talking to @DavidFair a logical and extensible way of accomplishing this would be to introduce a new key in the new TOML User Files, for example:
```
[instrument]
reduction_type = string  #where string = "SANS" / "SESANS" / "POLSANS"
```
This approach could, in the (further) future, then be extended to Focussing-SANS (which is outside the scope of this issue).

DavidFair, 2022-03-16:
On the reduction side it looks fairly simple the workflow would need to be adapted like so:

- Normalise by incident beam
- Normalise by transmission
- Convert to wavelength (for both cases)
- Normalise by supermirror / H3 cell.
- Rest of the reduction

For the super mirror we simply need to use a calibration run
For the H3 cell we fit the measured polarisation (which is captured multiple times), using [ExpDecay](https://docs.mantidproject.org/nightly/fitting/fitfunctions/ExpDecay.html) for the function `pHe(t) = pHe(t0) * exp(-t+t0/y)`, where y the relaxation time of the gas (this might need to be a user input, or an inst parameter depending how constant it is).
We then normalise by the value at `t/2` and our wavelength data.

From a UX point of view it might be worth having the multiple runs encapsulated in a separate page, rather than cluttering the main table. 
But I like Steve's idea to toggle between "Normal" SANS, Half polarised and Full Polarisation modes in the main interface above the runs table since it makes it trivial to see which mode were operating in without adding clutter.

dehoni, 2022-10-06:
The attached file [3HeMantidDataReduction_BetaVersion1D+2D_ZOOM+Larmor_combined_normalised.py.txt](https://github.com/mantidproject/mantid/files/9726406/3HeMantidDataReduction_BetaVersion1D%2B2D_ZOOM%2BLarmor_combined_normalised.py.txt) is a python script that contains two main parts: 1) `polarisation_parameter()`, which extract the polarisation parameters of the beamline (flipper, and polariser) and of the analyser cell and `calc_3He_parameters()` fitting the time dependence of the analysing power in case, one uses a 3He analyser for polarisation analysis.
2) Creating wavelength slices, performing (wavelength dependent) polarisation corrections of the 4 spin-resolved states with reduction matrixes, and merging data back together. 
The reduction is set up for Larmor and Zoom, and one can choose either correction of the 2D detector images or 1D scattering cross section. For the beamtime on Larmor-as the instrument does not have a beamstop monitor unlike ZOOM- we measured the attenuated, direct beam on the detector. The user only has to set up the correct list for transmission measurements through the 3He cell over the time, depolarised 3He cell to obtain the opaqueness of the 3He cell, empty direct beam and transmission/absorption of sample for normalisation, and -of course- scattering measurements.
Parameters like which monitor to use, and usable wavelength range should be ideally extracted from the user file, but I could not quite figure out how MANTID is doing it behind the scenes.
In general the code needs further cleanup and generalisation, a lot of the comments are related to me puzzling how MANTID works. There are several comments in the code where issues or future features are mentioned. I am sure there are more elegant ways than this crude patchy script, but it is a stumbling start.
[ISISCommandInterface_tweaked.py.txt](https://github.com/mantidproject/mantid/files/9726410/ISISCommandInterface_tweaked.py.txt)
Half-polarised SANS (with only incoming polarised beam) would use a similiar reduction without the additional calculations and matrixes for the 3He analyser appended.

dehoni, 2023-04-05:
The updated overview document for Polarization Analysis of TOF SANS contains now a basic workflow, with detailed function descriptions and some code examples in https://stfc365-my.sharepoint.com/:w:/g/personal/dirk_honecker_stfc_ac_uk/Eb5G595cOwVKqvJJByUpL2EBu-PUyZhuMXt2rf5YTiXs9Q?e=hii03t 
The general workflow scheme consists of two separate parts: Part A:calibration of the instrument polarisation (polariser, flipper, 3He gas polarisation, and automatic finding and assigning in which order the spin states were measured), and Part B: Polarisation correction on spin-resolved scattering data from a sample. 
The code is a first start needing some more work. The goal would be to have a code documentation with nicely isolated testable functions with expected input and output and finally to transform the hacked script into sustainable algorithms in Mantid.

rbauststfc, 2023-09-20:
The epic for implementing a Polarised SANS reduction has now been approved. The following are individual issues to capture the steps described in the epic. We will need to add details to these as part of requirements gathering and it may be that some issues are combined or removed as implementation decisions take shape:

- https://github.com/mantidproject/mantid/issues/36137
- https://github.com/mantidproject/mantid/issues/36138
- https://github.com/mantidproject/mantid/issues/36139
- https://github.com/mantidproject/mantid/issues/36140
- https://github.com/mantidproject/mantid/issues/36141
- https://github.com/mantidproject/mantid/issues/36142
- https://github.com/mantidproject/mantid/issues/36143
- https://github.com/mantidproject/mantid/issues/36144
- https://github.com/mantidproject/mantid/issues/36145
- https://github.com/mantidproject/mantid/issues/36146
- https://github.com/mantidproject/mantid/issues/35875
- https://github.com/mantidproject/mantid/issues/36147
- https://github.com/mantidproject/mantid/issues/36148

closed 2025-05-16 by github-actions[bot]