mantidproject/mantid #33645: LARMOR warning instead of crash

AnthonyLim23, 2022-03-10:
Raised by Robert

When loading run LARMOR 60202 Mantid crashes. This seems to be because it only contains log data (the raw file is empty and the log file is populated). - This happens in the load event nexus as it expects data, but it is empty. 

They have since said:

A warning would certainly be better than a full on crash with no error messages.

Obviously the ideal would be that the file would be loaded somehow so that the logs can be looked at but that would be a low priority.
I can use h5py and extract things by hand as this doesn’t happen very often.

I suggest for this issue we just prevent the crash.

AnthonyLim23, 2022-03-10:
@gemmaguest and @thomashampson I have got this from a user.

gemmaguest, 2022-03-11:
@thomashampson it's crashing because it's attempting to access values from an empty vector [here](https://github.com/mantidproject/mantid/blob/main/Framework/DataHandling/src/LoadEventNexus.cpp#L464). We could fix it by throwing an error if the vector is empty but maybe we should be checking earlier on if the raw file is empty?

thomashampson, 2022-03-14:
> @thomashampson it's crashing because it's attempting to access values from an empty vector [...]

Checking for an empty file early on sounds like the best solution.

thomashampson, 2022-10-20:
Has this been fixed already?