mantidproject/mantid #33659: Final tasks for release of SANS TOML v1

smk78, 2022-03-11:
This issue captures those remaining tasks necessary for moving to v1 of SANS TOML User Files identified during the review of https://github.com/mantidproject/mantid/pull/33065# and subsequently discussed with @DavidFair , @rmdalgliesh & Sarah.

Currently the monitors to be used in the reduction can be set in [instrument.configuration] _and_ in [normalisation] and [transmission]. This not only runs the risk of confusion for Users/Instrument Scientists, but could also lead to bad data depending on which location assumes precedence.

Agreed:
  
  - [ ] to _discontinue_ the use of selected_monitor in _both_ [normalisation] and [transmission] and to only select the monitors in [instrument.configuration].
  - [ ] to _change_ the data type of norm_monitor and trans_monitor in [instrument.configuration] from numeric (= 2, = 3) to string (= "M2", = "M3").
  - **not** to facilitate in TOML using the existing (but very, very, rarely used) code permitting reductions with a _different_ normalisation monitor for the transmissions; ie, the current behaviour in 6.3.0 is not to be changed.
  - [ ] _to facilitate_ in TOML the use of the existing (but unknown!) code for using a ROI for transmissions instead of a single monitor; ie, to implement what was TRANS/ROI=roi_file.xml and TRANS/MASK=mask_file.xml.
  
    This is to be done by allowing:

    [instrument.configuration]
      trans_monitor = "ROI"

    [transmission.ROI]
      file = "roi_file.xml"

    (Note that TRANS/MASK operates as an anti-ROI, but Mantid mask creation is flexible enough to allow both use cases in one file)
  - **not** to rename anything that was transmission.monitor as transmission.detector.

smk78, 2022-03-11:
Example User File conforming to the above. See #################### for changes.
[USER_SANS2D_as_agreed.toml.txt](https://github.com/mantidproject/mantid/files/8233560/USER_SANS2D_as_agreed.toml.txt)

DavidFair, 2022-03-11:
> not to facilitate in TOML using the existing (but very, very, rarely used) code permitting reductions with a different normalisation monitor for the transmissions; ie, the current behaviour in 6.3.0 is not to be changed.

Note this would be relatively easy for future reference. We would simply need to:
- Add relevant field
- Change
```python
            selected_norm_monitor = self.get_val("norm_monitor", inst_config_dict)
            self.calculate_transmission.incident_monitor = self.get_mandatory_val(
                [self._get_normalisation_spelling(), "monitor", selected_norm_monitor, "spectrum_number"])
```
To point from `normalisation.monitor.Mx.spectrum_number` where `Mx` comes from `instrument.configuration` to the new overriding field, e.g:
```python
            selected_norm_monitor = self.get_val("norm_monitor", inst_config_dict)
            if get_val(["transmission", "different_norm_monitor"]:
                    selected_norm_monitor = get_val(["transmission", "different_norm_monitor"]

            self.calculate_transmission.incident_monitor = self.get_mandatory_val(
                [self._get_normalisation_spelling(), "monitor", selected_norm_monitor, "spectrum_number"])
``` 

And add a new test based on `test_transmission` to check the new overriding monitor settings are picked up

closed 2022-03-31 by martyngigg