mantidproject/mantid #33661: SmoothNeighbours algorithm not clearing old workspace.

thomashampson, 2022-03-14:
**Describe the bug**
When running the SmoothNeighbours algorithm and overwriting the workspace, memory usage will be double the input workspace size.

**To Reproduce**
Run the following:
```
LoadRaw(Filename='/archive/ndxwish/Instrument/data/cycle_18_4/WISH00043526.raw', OutputWorkspace='43526') # ~20GB RAM
SmoothNeighbours(InputWorkspace='43526', OutputWorkspace='43526', Radius=3) # ~40GB RAM
```
Memory usage will be > 400 GB.

**Expected behavior**
When the workspace is replaced the memory usage should be only that occupied by the workspace.

**Screenshots**

**Platform/Version (please complete the following information):**
 - OS: All
 - Mantid Version 6.3.0, not tested yet on older versions.
 -

RichardWaiteSTFC, 2022-03-14:
To clarify the memory gets cleared when you delete it but if you don't delete the workspace it still behaves like the workspace has 40Gb not 20GB

martyngigg, 2022-04-14:
So I have done some investigation and confirmed with a simple script outside of workbench that this is an issue. The following script demonstrates the issue:

```python
"""SmoothNeighbours appears to hold on to memory when replacing a workspace with
itself in v6.3
"""
import psutil

MB = 1. / (1024*1024)

def pprint_memory_stats(process: psutil.Process):
    mem_info = process.memory_info()
    print(f"Resident Memory (this process): {mem_info.rss*MB:.3f}MB")

proc = psutil.Process()
print("Starting...")
pprint_memory_stats(proc)

print("Importing mantid")
from mantid.simpleapi import *
pprint_memory_stats(proc)

print("Loading file")
SXD23767 = Load(Filename='~/Data/TrainingCourseData/SXD23767.raw', LoadMonitors='Exclude')
pprint_memory_stats(proc)

print("Smoothing neighbours")
SXD23767 = SmoothNeighbours(SXD23767)
pprint_memory_stats(proc)
DeleteWorkspace(SXD23767)
pprint_memory_stats(proc)
```

On macOS Monteray it produces the following output

```sh
$ PYTHONUNBUFFERED=1 PYTHONPATH=$PWD/bin python ~/Code/mantid-scripts/smoothneigbours-memory.py
Starting...
Resident Memory (this process): 10.145MB
Importing mantid
FrameworkManager-[Notice] Welcome to Mantid 6.3.20220413.1031
FrameworkManager-[Notice] Please cite: http://dx.doi.org/10.1016/j.nima.2014.07.029 and this release: http://dx.doi.org/10.5286/Software/Mantid
Resident Memory (this process): 192.652MB
Loading file
Load-[Notice] Load started
Load-[Notice] Load successful, Duration 2.78 seconds
Resident Memory (this process): 1378.105MB
Smoothing neighbours
SmoothNeighbours-[Notice] SmoothNeighbours started
SmoothNeighbours-[Notice] SmoothNeighbours successful, Duration 3.61 seconds
Resident Memory (this process): 3076.922MB
DeleteWorkspace-[Notice] DeleteWorkspace started
DeleteWorkspace-[Notice] DeleteWorkspace successful, Duration 0.00 seconds
Resident Memory (this process): 1807.598MB
```

martyngigg, 2022-04-14:
`SmoothNeighbours` holds the input workspace as a member variable of the algorithm. Adding an `AlgorithmManager.Instance().clear()` after the `DeleteWorkspace` above frees the allocated memory.

Looking through the `AlgorithmManager` code and history I've identified this [commit](/mantidproject/mantid/commit/6c6d959a75103434c0dd3097d70d634c657f670d) that is responsible for holding on to the `SmoothNeighbours` instance. This is confirmed by adding `Pause(5)` immediately before the `DeleteWorkspace` call:

```python
"""SmoothNeighbours appears to hold on to memory when replacing a workspace with
itself in v6.3
"""
from signal import pause
import psutil

MB = 1. / (1024*1024)

def pprint_memory_stats(process: psutil.Process):
    mem_info = process.memory_info()
    print(f"Resident Memory (this process): {mem_info.rss*MB:.3f}MB")

proc = psutil.Process()
print("Starting...")
pprint_memory_stats(proc)

print("Importing mantid")
from mantid.simpleapi import *
pprint_memory_stats(proc)

print("Loading file")
SXD23767 = Load(Filename='~/Data/TrainingCourseData/SXD23767.raw', LoadMonitors='Exclude')
pprint_memory_stats(proc)

print("Smoothing neighbours")
SXD23767 = SmoothNeighbours(SXD23767)
pprint_memory_stats(proc)
Pause(5)
DeleteWorkspace(SXD23767)
AlgorithmManager.Instance().clear()
pprint_memory_stats(proc)
```

produces output:

```sh
$ PYTHONUNBUFFERED=1 PYTHONPATH=$PWD/bin python ~/Code/mantid-scripts/smoothneigbours-memory.py
Starting...
Resident Memory (this process): 9.980MB
Importing mantid
FrameworkManager-[Notice] Welcome to Mantid 6.3.20220413.1031
FrameworkManager-[Notice] Please cite: http://dx.doi.org/10.1016/j.nima.2014.07.029 and this release: http://dx.doi.org/10.5286/Software/Mantid
Resident Memory (this process): 193.688MB
Loading file
Load-[Notice] Load started
Load-[Notice] Load successful, Duration 2.94 seconds
Resident Memory (this process): 1381.934MB
Smoothing neighbours
SmoothNeighbours-[Notice] SmoothNeighbours started
SmoothNeighbours-[Notice] SmoothNeighbours successful, Duration 3.38 seconds
Resident Memory (this process): 3076.227MB
Pause-[Notice] Pause started
Pause-[Notice] Pause successful, Duration 5.02 seconds
DeleteWorkspace-[Notice] DeleteWorkspace started
DeleteWorkspace-[Notice] DeleteWorkspace successful, Duration 0.00 seconds
Resident Memory (this process): 546.758MB
```

The upside of this is there is no actual leak and, as we have observed, running further algorithms does return the memory back, just a little later than expected. The change linked above was necessary to avoid a crash in code linked to handling algorithm finished notifications that were trying to use an object that had been deleted. I will think on a solution to this.

RichardWaiteSTFC, 2022-04-14:
@martyngigg - I was able to run out of memory by loading in other workspaces after running `SmoothNeighbours` - it didn't look like the RAM was ever freed up - is that consistent with what you have seen?

martyngigg, 2022-04-14:
> @martyngigg - I was able to run out of memory by loading in other workspaces after running `SmoothNeighbours` - it didn't look like the RAM was ever freed up - is that consistent with what you have seen?

If other algorithms were run quick enough then yes as the old algorithms are only freed > 5s after a previous algorithm has run but also after a new one is created. I'm currently thinking if we can change the way the notifications are handled.

RichardWaiteSTFC, 2022-04-14:
> If other algorithms were run quick enough then yes as the old algorithms are only freed > 5s after a previous algorithm has run but also after a new one is created. I'm currently thinking if we can change the way the notifications are handled.

I was running it manually using the algorithm GUI and I think it takes more then 5s for a WISH workspace to load (and i needed to load in 4) - so at least for me on IDAaaS (on that day!) the memory wasn't being freed, but it was a while ago I could be wrong.

martyngigg, 2022-04-14:
> I was running it manually using the algorithm GUI and I think it takes more then 5s for a WISH workspace to load (and i needed to load in 4) - so at least for me on IDAaaS (on that day!) the memory [...]

Thanks for the extra information. I'm pretty sure this is the issue but I'll test this case out too.

PR #33790 (2022-04-29): Clean internal variables inside SmoothNeighbours
**Description of work.**

The first 3 commits are cleanup, the final is a release note. The real fix is inside commit 4. Please see that commit message for more details.

**To test:**

The following script should so the memory usage go down after the `DeleteWorkspace` call. Before this fix it did not.

```python
"""SmoothNeighbours appears to hold on to memory when replacing a workspace with
itself in v6.3
"""
import psutil

MB = 1. / (1024*1024)

def pprint_memory_stats(process: psutil.Process):
    mem_info = process.memory_info()
    print(f" [... cut]
closed 2022-04-29 by DannyHindson