mantidproject/mantid #33774: Convolution on non-equidistant grids

darominski, 2022-04-19:
**Describe the bug**

An ILL scientific user has reported that convolutions in Mantid do not behave as they would expect. It seems that there is a strong dependence on the function domain, in particular, in case where the grid has non-equidistant spacing.

**To Reproduce**
Steps to reproduce the behavior:

1. Run the following script, observe that different grids lead to significant differences in convolutions:

```
# import mantid algorithms, numpy and matplotlib
from mantid.simpleapi import *
import matplotlib.pyplot as plt
import numpy as np

G = Gaussian(Height=1, Sigma=0.2)
L = Lorentzian(Amplitude=1, FWHM=0.5)

F = Convolution(G, L)

x1 = np.linspace(-2,2,100)
x2 = np.concatenate( (np.linspace(-2,-0.5,20), np.linspace(-0.5,2.0,100)) )
x3 = np.log(np.linspace(np.exp(-2), np.exp(2), 100))

plt.figure()
plt.plot( x1, G(x1), '--', label="Function 1")
plt.plot( x1, L(x1), '--', label="Function 2" )
plt.plot( x1, F(x1), '.-', label="Convolution on equidistant grid" )
plt.plot( x2, F(x2), '.-', label="Convolution on non-equidistant grid" )
plt.plot( x3, F(x3), '.-', label="Convolution on another non-equidistant grid" )
plt.legend()
plt.show()
```
 
**Expected behavior**

Convolution outcome does not depend on the input grid.

**Screenshots**

**Platform/Version (please complete the following information):**
 - OS: [e.g. Windows, RHEL 7, Ubuntu, macOS]
 - OS Version:
 - Mantid Version [e.g. 6.0.0]

**Additional context**

AnthonyLim23, 2022-04-27:
The convolution fit function does an [FFT] (https://docs.mantidproject.org/nightly/fitting/fitfunctions/Convolution.html) as part of the calculation. The issue here is that an FFT assumes that the x domain is equally spaced (its baked into the assumptions to make it a quicker). 

To get this to work you would need to not use a different numerical method (see https://scicomp.stackexchange.com/questions/593/how-do-i-take-the-fft-of-unevenly-spaced-data) which could be a significant drop in performance (a factor of 4 or 5 for 1D data).

closed 2022-07-26 by martyngigg

AnthonyLim23, 2022-11-16:
I needed to make a few changes to get the above script to run

```
# import mantid algorithms, numpy and matplotlib
from mantid.simpleapi import *
import matplotlib.pyplot as plt
import numpy as np

# import mantid algorithms, numpy and matplotlib
from mantid.simpleapi import *
import matplotlib.pyplot as plt
import numpy as np

G = Gaussian(Height=1, Sigma=0.2)
L = Lorentzian(Amplitude=1, FWHM=0.5)

def F(x):
    ff = Convolution(G, L)
    ss = CreateWorkspace(x,x)
    fit = Fit(Function=ff, 
        InputWorkspace=ss, Output='bob', OutputCompositeMembers=False, MaxIterations=0)
    ws = fit.OutputWorkspace

    return ws.readY(1)
    
x1 = np.linspace(-2,2,100)
x2 = np.concatenate( (np.linspace(-2,-0.5,20), np.linspace(-0.5,2.0,100)) )
x3 = np.log(np.linspace(np.exp(-2), np.exp(2), 100))

plt.figure()
#plt.plot( x1, G(x1), '--', label="Function 1")
#plt.plot( x1, L(x1), '--', label="Function 2" )
plt.plot( x1, F(x1), '.-', label="Convolution on equidistant grid" )
plt.plot( x2, F(x2), '.-', label="Convolution on non-equidistant grid" )
plt.plot( x3, F(x3), '.-', label="Convolution on another non-equidistant grid" )
plt.legend()
plt.show()
```

AnthonyLim23, 2022-11-16:
```
# import mantid algorithms, numpy and matplotlib
from mantid.simpleapi import *
import matplotlib.pyplot as plt
import numpy as np

# import mantid algorithms, numpy and matplotlib
from mantid.simpleapi import *
import matplotlib.pyplot as plt
import numpy as np
from scipy.interpolate import interp1d

G = Gaussian(Height=1, Sigma=0.2)
L = Lorentzian(Amplitude=1, FWHM=0.5)

def F(x):
    ff = Convolution(G, L)
    ss = CreateWorkspace(x,x)
    fit = Fit(Function=ff, 
        InputWorkspace=ss, Output='bob', OutputCompositeMembers=False, MaxIterations=0)
    ws = fit.OutputWorkspace

    return ws.readY(1)
    
 
def NUcon(x, y1, y2):
    """
    Resample onto uniform grid using the smallest interval 
    """
    start = x[0]
    end = x[-1]
    dx = np.abs(np.min(x[1:] - x[:-1]))
    N = int((end-start)/dx)
    new_x = np.linspace(start, end, N)
    f1 = interp1d(x, y1, bounds_error=False, kind='cubic')
    new_y1 = f1(new_x)
    f2 = interp1d(x, y1, bounds_error=False, kind='cubic')
    new_y2 = f2(new_x)
    return new_x, np.convolve(new_y1, new_y2, mode='same')/np.sum(new_y2)
    
x1 = np.linspace(-2,2,100)
print(x1[1] - x1[0])
x2 = np.concatenate( (np.linspace(-2,-0.5,20), np.linspace(-0.55,2.0,100)) )
x3 = np.log(np.linspace(np.exp(-2), np.exp(2), 100))

plt.figure()
plt.plot( x1, F(x1), 'k-', label="Convolution on equidistant grid" )

plt.plot( x2, F(x2), 'b--', label="Convolution on non-equidistant grid" )
plt.plot( x3, F(x3), 'y--', label="Convolution on another non-equidistant grid" )

y1 = G(x2)
y2 = L(x2)
x, y = NUcon(x2, y1, y2)
plt.plot( x, y,  'g-', label="Convolution on non-equidistant grid v2")

y1 = G(x3)
y2 = L(x3)
x, y = NUcon(x3, y1, y2)
plt.plot( x, y, 'r--', label="Convolution on another non-equidistant grid v2")

plt.legend()
plt.show()
```