mantidproject/mantid #33777: Indirect Data Reduction Moments tab does not work for OSIRIS Data

Harrietbrown, 2022-04-20:
**Original reporter:** @SpencerHowells

**Describe the bug**
In the moments tab in the Indirect Data if you attempt to run with OSIRIS data the output is all NaN. This is most likly coming from an issue in the `SofQWMoments` algorithm.

**To Reproduce**
1. open IndirectDataReduction
2. go to the Moments tab
3. Load some OSIRIS sqw data
4. Click run
5. Check the moments workspace and see that it is all nan.

AnthonyLim23, 2022-04-27:
It looks like the issue is in the integration alg. The input data looks ok to me. I have stripped the algorithm done to the below script:

```

def do(name, m1, m2):

        input_ws = name
        #workflow_prog.report('Cropping Workspace')
        crop_alg = AlgorithmManager.create("CropWorkspace")
        crop_alg.setProperty("InputWorkspace", name)
        crop_alg.setProperty("XMin", m1)
        crop_alg.setProperty("XMax", m2)
        crop_alg.setProperty("OutputWorkspace", "crop")
        crop_alg.execute()
        #mtd.addOrReplace("crop", crop_alg.getProperty("OutputWorkspace").value)
        
        # calculate delta x
        #workflow_prog.report('Converting to point data')
        input_ws = "crop"
        convert_point_alg = AlgorithmManager.create("ConvertToPointData")
        convert_point_alg.setProperty("InputWorkspace", input_ws)
        convert_point_alg.setProperty("OutputWorkspace", "point")
        convert_point_alg.execute()
        
        input_ws = "point"
        convert_hist_alg =  AlgorithmManager.create("ConvertToHistogram")
        convert_hist_alg.setProperty("InputWorkspace", input_ws)
        convert_hist_alg.setProperty("OutputWorkspace", "hist")
        convert_hist_alg.execute()

        ws = mtd["crop"]
        
        data = []
        for spec in range(ws.getNumberHistograms()):
            x_values = ws.readX(spec)
            y_values = ws.readY(spec)
            I = 0.0
            for j in range(len(y_values)):
                I += (x_values[j+1]-x_values[j])*y_values[j]
            print(I)

        integration_alg =  AlgorithmManager.create("Integration")
        integration_alg.setProperty("InputWorkspace", input_ws)# convert_hist_alg.getProperty("OutputWorkspace").value)
        integration_alg.setProperty("OutputWorkspace", "int")
        integration_alg.execute()

       

Load(Filename='osiris144076_graphite002_sqw.nxs', OutputWorkspace='osiris144076_graphite002_sqw')
Rebin(InputWorkspace='osiris144076_graphite002_sqw', OutputWorkspace='osiris144076_graphite002_sqw', Params='-0.5,0.005,0.5')
do('osiris144076_graphite002_sqw',-.4,.4)

```

AnthonyLim23, 2022-04-27:
I think its coming from this set of changes: https://github.com/mantidproject/mantid/commit/63f2ca0a3eed4f67769f31ff173d41f8baf6db6d

mducle, 2022-04-28:
Ok... so the issue is [XDataConverter](https://github.com/mantidproject/mantid/blob/03deeb89254ec4289edb8771e0188c2090a02f32/Framework/Algorithms/src/XDataConverter.cpp#L69-L76) which is used by `ConvertToPointData` (and `ConvertToHistogram`) to do the actual conversion. I basically ignores the fractions in `RebinnedOutput`. A fix would be to change lines 64-80 in `XDataConverter.cpp` to something like:

```c++
  bool isRebinnedWorkspace = inputWS->id() == "RebinnedOutput";

  Progress prog(this, 0.0, 1.0, numSpectra);
  PARALLEL_FOR_IF(Kernel::threadSafe(*inputWS, *outputWS))
  for (int i = 0; i < int(numSpectra); ++i) {
    PARALLEL_START_INTERRUPT_REGION

    // Copy over the Y and E data
    outputWS->setSharedY(i, inputWS->sharedY(i));
    outputWS->setSharedE(i, inputWS->sharedE(i));
    if (isRebinnedWorkspace) {
      outputWS->setF(i, inputWS->dataF(i));
    }
    setXData(outputWS, inputWS, i);
    if (inputWS->hasDx(i)) {
      outputWS->setSharedDx(i, inputWS->sharedDx(i));
    }
    prog.report();

    PARALLEL_END_INTERRUPT_REGION
  }
  PARALLEL_CHECK_INTERRUPT_REGION

  // Store the output
  setProperty("OutputWorkspace", outputWS);
}
```

It might make it run a bit slower because I think `setF` has to copy the data rather than just a pointer like `setSharedY` - there isn't a `setSharedF` (or `sharedF`) method as far as I can tell... 

The alternative is to change the `SofQWMoments` algorithm to not use `ConvertToPointData` (and then `ConvertToHistogram` to convert it back) because the original input workspace is already histogram data; it seems to me that the only reason to convert to point is to get the X-values. This could instead be done directly in the Python without converting the workspace; the multiplications after that conversion should still work ok (I think!).

So the changes would be to lines 69-84 of `SofQWMoments.py` to something like:

```python
        x_data = np.asarray(mtd[input_ws].readX(0))
        x_data = (x_data[1:] + x_data[:-1]) / 2.0
        workflow_prog.report('Creating temporary data workspace')
        x_workspace = "__temp_sqw_moments_x"
        create_alg = self.createChildAlgorithm("CreateWorkspace", enableLogging=False)
        create_alg.setProperty("DataX", x_data)
        create_alg.setProperty("DataY", x_data)
        create_alg.setProperty("UnitX", "DeltaE")
        create_alg.setProperty("OutputWorkspace", x_workspace)
        create_alg.execute()
        mtd.addOrReplace(x_workspace, create_alg.getProperty("OutputWorkspace").value)
```

PR #33798 (2022-05-06): Remove convert to point data from SofQWMoments
**Description of work.**

This PR removes a use of the convert to point data from the SofQWMoments algorithm. Fixing an issue that stopped it working with OSIRIS data.

**Report to:** @SpencerHowell

**To test:**

Run the following and check that the output `_moments` workspace is not NaN.

```
from mantid.simpleapi import *

ISISIndirectEnergyTransferWrapper(InputFiles=r'\\isis.cclrc.ac.uk\inst$\ndxosiris\instrument\data\cycle_20_2\OSIRIS00144076.nxs', Instrument='OSIRIS', Analyser='graphite', Reflection='002', Efixed=1.845, SpectraRange='963,100 [... cut]
closed 2022-05-06 by martyngigg