mantidproject/mantid #33779: Whitespaces in data path

darominski, 2022-04-21:
**Describe the bug**
There is an issue in validation when whitespaces are present in the paths or names of the data, which prevents loading of such data. The issue has been observed only when full paths are given to a loader. In such a case, the data does not pass validation and the error message contains information that the file cannot be found, in which all whitespaces are removed, making the path incorrect. This issue does not arise when only file names are given, without full paths.

**To Reproduce**

1. Create a directory or files that have a whitespace, e.g. create a couple of workspaces using `CreateSampleWorkspace` and save them in preferred location.
2. Try to load these files using e.g. `LoadAndMerge` algorithm, observe the validation issue. The path or name mentioned in validation issue is not consistent with actual name of the input to the `Filename` property.

**Expected behavior**

It should be possible to load data even if the path or the filename contains (an unfortunate) white space.

**Screenshots**

**Platform/Version (please complete the following information):**
 - OS: observed on Ubuntu 18.04, Windows 10
 - OS Version:
 - Mantid Version 6.3

**Additional context**

sf1919, 2022-04-21:
Possibly related to #30913 ?

darominski, 2022-04-21:
> Possibly related to #30913 ?

It does seem so. My search for similar issues was then too narrow. I wonder if the proposed solution also could address the issue with different delimiters mentioned in #30913

closed 2022-05-04 by martyngigg