mantidproject/mantid #33833: Manual Testing Workbench Group 1- 4

Harrietbrown, 2022-05-06:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

https://developer.mantidproject.org/Testing/Core/Core.html

martyngigg, 2022-05-11:
I'm just noting I opened #33869 for a missing icon

DanielMurphy22, 2022-05-11:
Sorry to hijack, but feels most appropriate place to mention the font change on tar install on Ubuntu: https://github.com/mantidproject/mantid/issues/33875

ConorMFinn, 2022-05-12:
[image]

RichardWaiteSTFC, 2022-05-12:
Changing colorbar   log scale in figure options works (sets min limit to 0.001 if 0) - but if you double click on a number in a colorbar and set log it behaves differently
[image]

thomashampson, 2022-05-12:
Waterfall plot (SXD23745) - plotting spectra with errorbars, set cap size to something other than zero, then adjust offset amount (of Y offset)

Stack Trace:
       Traceback (most recent call last):
 File "C:\MantidNightlyInstall\bin\mantidqt\widgets\waterfallplotoffsetdialog\presenter.py", line 27, in <lambda>
   self.view.y_offset_spin_box.valueChanged.connect(lambda: self.update_y_offset())
 File "C:\MantidNightlyInstall\bin\mantidqt\widgets\waterfallplotoffsetdialog\presenter.py", line 55, in update_y_offset
   self.ax.update_waterfall(self.view.get_x_offset(), new_y)
 File "C:\MantidNightlyInstall\bin\mantid\plots\mantidaxes.py", line 1101, in update_waterfall
   self.lines.add_line(cap)
AttributeError: 'ArtistList' object has no attribute 'add_line'

SilkeSchomann, 2022-05-13:
Testing instructions for Install/Uninstall are not applicable for Windows conda.

robertapplin, 2022-05-13:
Clicking the print button on a plot seems to freeze for a long time (20 seconds) and sometimes says 'Not Responding', before finally opening a print dialog
[image]

robertapplin, 2022-05-13:
When I add a vertical marker to a plot, and then zoom out of the plot, the label for the marker becomes detached from its line (as if it uses a different coordinate system)
[image]

SilkeSchomann, 2022-05-13:
I was running the example script from https://docs.mantidproject.org/nightly/tutorials/python_in_mantid/plotting/02_scripting_plots.html#example-script
and clicking on Plot Options. 

[image]

RichardWaiteSTFC, 2022-05-13:
Problem with normalisation when you plot two spectra and set the errorbars to be visible on one then change normalisation to None - it looks like it doesn't change the normalisation of the spectrum that errrobars were added to.
[image]

RichardWaiteSTFC, 2022-05-13:
Plot to script to clipboard/file doesn't plot markers (vertical or horizontal) on IDAaaS

SilkeSchomann, 2022-05-13:
I found this instruction in https://www.mantidproject.org/ScriptRepository: If you upload your scripts by the 2nd of August we promise you we will help you migrating them to the new version of python

It probably needs an update ;-)

RichardWaiteSTFC, 2022-05-13:
Right-click on script tab in editor > Show in Explorer doesn't work on IDAaaS

RichardWaiteSTFC, 2022-05-13:
Trying to install script_repo on IDAaaS causes hard crash (with no stack trace)

robertapplin, 2022-05-13:
Create a ragged workspace using this script:
```
from mantid.simpleapi import *
import matplotlib.pyplot as plt
import numpy as np
ws = CreateWorkspace([0,1,2,3,4,2, 3,4,5,6],[0,1,2,3,4,3,4,5,6,7], NSpec=2)
min = 1
max = 6
for j in range(0,2):
    x =[]
    y =[]
    data = ws.readY(j)
    k=0
    for v in ws.readX(j):
        if v >= min and v<=max:
            x.append(v)
            y.append(data[k])
        k+=1
    tmp = CreateWorkspace(x,y)
    ConjoinWorkspaces(ws, tmp, CheckOverlapping=False)
```
Open the ragged workspace with `Show Data`. Right click a row and do `Copy Spectrum to table`. There is an error:
```
Traceback (most recent call last):
  File "C:\MantidNightlyInstall\bin\mantidqt\widgets\workspacedisplay\matrix\presenter.py", line 104, in action_copy_spectrum_to_table
    num_rows = ws.blocksize()
RuntimeError: blocksize undefined because size of histograms is not equal
```

ConorMFinn, 2022-05-13:
All issues created