mantidproject/mantid #33834: Manual Testing Muon

Harrietbrown, 2022-05-06:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

See instructions at: https://developer.mantidproject.org/Testing/MuonAnalysis_test_guides/index.html
https://developer.mantidproject.org/Testing/MuonInterface/MuonTesting.html

AnthonyLim23, 2022-05-12:
In ALC doc:

change from marker to add section

DanielMurphy22, 2022-05-12:
To see all of the data press the select plot data button and select all of the rows in the pop-up

AnthonyLim23, 2022-05-12:
MUSR wrong global

AnthonyLim23, 2022-05-12:
psi change from 0.007 to 0.006

AnthonyLim23, 2022-05-12:
On EMU move the external plot line

Pressing the external plot button caused a crash

AnthonyLim23, 2022-05-12:
on superconducting remove the bit on step through the plots

AnthonyLim23, 2022-05-12:
FDA unit change is not working

AnthonyLim23, 2022-05-12:
Maxent group name needs updating

AnthonyLim23, 2022-05-12:
HIFI need to say to say to change selected groups

AnthonyLim23, 2022-05-13:
doc issue is https://github.com/mantidproject/mantid/issues/33904

closed 2022-05-13 by AnthonyLim23