mantidproject/mantid #33835: Manual Testing TOFConverter

Harrietbrown, 2022-05-06:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

Make sure that inputs and outputs work sensibly, stress test with some bad inputs (e.g. letters in a numeric input)
https://docs.mantidproject.org/interfaces/utility/TOF%20Converter.html

RichardWaiteSTFC, 2022-05-12:
Covnerting between Q and d-should not need the scattering angle (Q = 2pi/d)
[image]

RichardWaiteSTFC, 2022-05-12:
It might be nice not to allow the user to convert to the same unit , e.g. TOF -> TOF
In the case you do allow it you shouldn't need the path length!
[image]

RichardWaiteSTFC, 2022-05-12:
TOF converter can use a single valued workspace, but it doesn't look like it can!

RichardWaiteSTFC, 2022-05-12:
@Harrietbrown @DanielMurphy22 - are any of these important enough to make issues for?

DanielMurphy22, 2022-05-12:
I think the answer is no. It feels like a low priority nice to have that in the past would have been an induction issue, but maybe isn't even worth it. @martyngigg do you think these problems are worth an induction issue?

ConorMFinn, 2022-05-13:
Induction issue made for what we found at #33898

closed 2022-05-13 by ConorMFinn