mantidproject/mantid #33837: Manual Testing Engineering Diffraction

Harrietbrown, 2022-05-06:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

https://developer.mantidproject.org/Testing/EngineeringDiffraction/EngineeringDiffractionTestGuide.html

DannyHindson, 2022-05-11:
1) plots on Calibration output aren't quite the same as in instructions (data different)
2) Martyn only got all banks .prm with an error reported from shutil
3) if change default save location it doesn't appear to be taken into account
4) step 3) of Focus. Plot doesn't have bank 2 on it
5) screenshot of step 5) is out of date, has 193749 in instructions
6) Test 3 - need to select "Create New Calibration" before cropping instructions available
7) Change "custom spectra" to "Crop" to spectra in instructions
8) There are now two texture groupings - only one in instructions
9) default value in Load box of fitting tab seems to have all previous focusses in it rt just the last one (eg we focussed texture then 2 banks) and it loaded them all
10) undock of plot has left the toolbar behind
11) Test 5. Step 1 - workspace is deleted from ADS on Remove. Is this correct?
12) Test 5. Step 3 - didn't see any calls to AverageLogData at any point
13) Test 6. what values of Niter etc should give a new plot??
14) Test 7. Removing run shut fit property browser down (but did remove run!)
15) Test 8. If we change the primary log, should it kick in immediately or do you need to close UI or even Mantid down?

DannyHindson, 2022-05-12:
Have opened three issues to cover the definite problems identified:
- https://github.com/mantidproject/mantid/issues/33877
- https://github.com/mantidproject/mantid/issues/33878
- https://github.com/mantidproject/mantid/issues/33879

Will discuss some of the "maybes" with Richard at some point but this testing issue can be closed