mantidproject/mantid #33841: Manual Testing ISIS SANS

Harrietbrown, 2022-05-06:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

https://developer.mantidproject.org/Testing/SANSGUI/SANSGUITests.html

robertapplin, 2022-05-11:
The `MaskFile.toml` in the training course data appears to be out of date

robertapplin, 2022-05-11:
Note on IDAaaS instructions that instrument viewer doesn't open with 'Full3D' on purpose when opened from SANS

gemmaguest, 2022-05-20:
> The `MaskFile.toml` in the training course data appears to be out of date

I've now updated this

gemmaguest, 2022-05-20:
- CanSAS is not disabled for 2D reductions. There are a few issues related to this so I've extended this issue to deal with them:  #30475
- The other main issue we found was #33962

closed 2022-05-20 by gemmaguest