mantidproject/mantid #33842: Manual Testing Live Data ISIS

Harrietbrown, 2022-05-06:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

https://developer.mantidproject.org/Testing/LiveData/LiveDataTests.html

DannyHindson, 2022-05-11:
1) instructions should mention to click "run" button on StartLiveData screen
2) crash on show data on "live" workspace in testing instructions section "Live Histogram Data". Instrument=ISIS_Histogram
3) also crashes on show data on "live" workspace on "Live event pre and post processing". Instrument=ADARA_FakeEvent

Stack traces for 2) and 3):

Name: Not provided  Email: Not provided
       Additional text:
       Error from show data on live data (manual test)
       Stack Trace:
       Traceback (most recent call last):
 File "/opt/mantidworkbenchnightly/lib/python3.8/site-packages/mantidqt/widgets/workspacedisplay/matrix/presenter.py", line 71, in action_replace_workspace
   if self.model.workspace_equals(workspace_name):
 File "/opt/mantidworkbenchnightly/lib/python3.8/site-packages/mantidqt/widgets/workspacedisplay/matrix/model.py", line 52, in workspace_equals
   return workspace_name == self._ws.name()
RuntimeError: Variable invalidated, data has been deleted.

DannyHindson, 2022-05-12:
Opened issue https://github.com/mantidproject/mantid/issues/33885 for crash

closed 2022-05-12 by martyngigg