mantidproject/mantid #33843: Manual Testing Mantid Basics Course

Harrietbrown, 2022-05-06:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

* Check the Qt-help docs, built into Workbench, against the online documentation 
* Open up the basic course: http://docs.mantidproject.org/tutorials/mantid_basic_course/index.html#mantid-basic-course
* Check that the information and exercises are up to date and work in the Workbench release candidate

martyngigg, 2022-05-12:
When testing the "Loading & Display Data" section. Plotting the colorfill of MAR1060 gives 2 warnings:
```python
Scale is set to logarithmic so non-positive min value has been changed to 0.0001.
Support for case-insensitive scales is deprecated since 3.5 and support will be removed two minor releases later.
```

The first is expected as the min scale value is too small.

The second deprecation warning has come from matplotlib 3.5 and should be suppressed/fixed.

martyngigg, 2022-05-12:
When testing the sliceviewer section, loaded MAR11060, selected "Region selector" button and a warning:
```python
The 'drawtype' parameter of __init__() was deprecated in Matplotlib 3.5 and will be removed two minor releases later. If any parameter follows 'drawtype', they should be passed as keyword, not positionally.
```

martyngigg, 2022-05-12:
Another matplotlib 3.5 deprecation warning. Unsure on what we did to reproduce but it stated the `get_window_title` function is deprecated. It is probably worth searching the codebase for its usage.

DannyHindson, 2022-05-12:
Few issues found on the Fitting Data section (https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/fitting_data/index.html#fitting-data):
- ~~on the "More Complex Case" page it talks about being able to change the height of the "central solid red line" after adding a Gaussian peak to a plot of HRP39182. We weren't able to alter the height~~ (just tried and this actually works for me)
- Whenever "Plot Guess" is run the plot is zoomed right out again. Similarly loading a previously saved model also zooms right out
- **crash on selecting "Add other function", "UserFunction" to a plot with a guess present on the plot:**

```
Name: Harriet  Email: Not provided
       Additional text:
       Added UserFunction to a plot which already had a guess plotted.
       Stack Trace:
       Traceback (most recent call last):
 File "/opt/mantidworkbenchnightly/lib/python3.8/site-packages/mantidqt/widgets/fitpropertybrowser/fitpropertybrowser.py", line 541, in function_changed_slot
   self.plot_interaction_manager.update_guess()
 File "/opt/mantidworkbenchnightly/lib/python3.8/site-packages/mantidqt/widgets/fitpropertybrowser/fitpropertybrowserplotinteraction.py", line 195, in update_guess
   self.update_guess_all()
 File "/opt/mantidworkbenchnightly/lib/python3.8/site-packages/mantidqt/widgets/fitpropertybrowser/fitpropertybrowserplotinteraction.py", line 234, in update_guess_all
   line = self._plot_guess_workspace(ws_name, fun, out_ws_name, color=color)
 File "/opt/mantidworkbenchnightly/lib/python3.8/site-packages/mantidqt/widgets/fitpropertybrowser/fitpropertybrowserplotinteraction.py", line 271, in _plot_guess_workspace
   out_ws = self.evaluate_function(workspace_name, function, output_workspace_name)
 File "/opt/mantidworkbenchnightly/lib/python3.8/site-packages/mantidqt/widgets/fitpropertybrowser/fitpropertybrowserplotinteraction.py", line 139, in evaluate_function
   alg.execute()
ValueError: Error evaluating function: Unexpected end of expression at position 0
```

DannyHindson, 2022-05-12:
One minor documentation thing from the Connecting Data To Instruments section (https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/connecting_data_to_instruments/03_investigating_data.html#investigating-data):
- the toolbar on the Instrument Viewer Pick tab now has one extra button that's not in the screenshot\documentation which is "Select Whole Instrument"

DannyHindson, 2022-05-12:
Another minor documentation thing from the Displaying 2D data section (https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/loading_and_displaying_data/04_displaying_2D_data.html#displaying-2d-data):
- the screenshots in in the instructions show the units at the top of the colourfill plots. The workspace name is there instead in workbench now with the units being next to the legend:

[image]

closed 2022-05-16 by Harrietbrown

DannyHindson, 2022-05-16:
Issues raised:
- Zoom issue when plotting guess had already been raised by Richard: https://github.com/mantidproject/mantid/issues/33909
- https://github.com/mantidproject/mantid/issues/33925