mantidproject/mantid #33844: Manual Testing M-slice

Harrietbrown, 2022-05-06:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

https://developer.mantidproject.org/Testing/MSlice/MSliceTestGuide.html

SilkeSchomann, 2022-05-09:
It would be good to test the latest version of MSlice as a conda package as well. The latest available nightly build can be found here: https://github.com/mantidproject/mslice/actions/runs/2158705974

For testing it needs to be unzipped and installed in a local conda channel:

1. Create a directory, for example _mycondachannel_
2. In this new directory, add a subdirectory called _noarch_
3. Copy the .tar.bz2 file for MSlice into _noarch_
4. Run `conda index mycondachannel`
5. Run `conda install -c C:\path\to\mycondachannel mslice`

ajjackson, 2022-05-12:
On Mac (once I actually followed the directory-name instructions correctly!) I found that

`mamba install -c /full/path/to/mslicecondachannel -c mantid/label/nightly mslice`

did the trick. Without the Mantid nightly channel it couldn't find that dependency.

thomashampson, 2022-05-12:
Open a plot and double click the colour bar to open the dialog. Set the max to something negative, AND select logarithmic. Click OK. This casues a crash. Change either of those things individually is handled nicely, but together it causes a problem.

<details>
  <summary>Stack trace</summary>

```
Traceback (most recent call last):
  File "C:\Users\dsc49661\AppData\Local\mambaforge\envs\mslice_new\lib\site-packages\matplotlib\cbook\__init__.py", line 287, in process
    func(*args, **kwargs)
  File "C:\Users\dsc49661\AppData\Local\mambaforge\envs\mslice_new\lib\site-packages\matplotlib\colorbar.py", line 551, in update_normal
    self.draw_all()
  File "C:\Users\dsc49661\AppData\Local\mambaforge\envs\mslice_new\lib\site-packages\matplotlib\colorbar.py", line 578, in draw_all
    self._process_values()
  File "C:\Users\dsc49661\AppData\Local\mambaforge\envs\mslice_new\lib\site-packages\matplotlib\colorbar.py", line 1144, in _process_values
    b = self.norm.inverse(b)
  File "C:\Users\dsc49661\AppData\Local\mambaforge\envs\mslice_new\lib\site-packages\matplotlib\colors.py", line 1554, in inverse
    raise ValueError("Invalid vmin or vmax")
ValueError: Invalid vmin or vmax
Traceback (most recent call last):
  File "C:\Users\dsc49661\AppData\Local\mambaforge\envs\mslice_new\lib\site-packages\matplotlib\cbook\__init__.py", line 287, in process
    func(*args, **kwargs)
  File "C:\Users\dsc49661\AppData\Local\mambaforge\envs\mslice_new\lib\site-packages\matplotlib\colorbar.py", line 551, in update_normal
    self.draw_all()
  File "C:\Users\dsc49661\AppData\Local\mambaforge\envs\mslice_new\lib\site-packages\matplotlib\colorbar.py", line 578, in draw_all
    self._process_values()
  File "C:\Users\dsc49661\AppData\Local\mambaforge\envs\mslice_new\lib\site-packages\matplotlib\colorbar.py", line 1144, in _process_values
    b = self.norm.inverse(b)
  File "C:\Users\dsc49661\AppData\Local\mambaforge\envs\mslice_new\lib\site-packages\matplotlib\colors.py", line 1554, in inverse
    raise ValueError("Invalid vmin or vmax")
ValueError: Invalid vmin or vmax
Traceback (most recent call last):
  File "C:\Users\dsc49661\AppData\Local\mambaforge\envs\mslice_new\lib\site-packages\matplotlib\cbook\__init__.py", line 287, in process
    func(*args, **kwargs)
  File "C:\Users\dsc49661\AppData\Local\mambaforge\envs\mslice_new\lib\site-packages\matplotlib\colorbar.py", line 551, in update_normal
    self.draw_all()
  File "C:\Users\dsc49661\AppData\Local\mambaforge\envs\mslice_new\lib\site-packages\matplotlib\colorbar.py", line 578, in draw_all
    self._process_values()
  File "C:\Users\dsc49661\AppData\Local\mambaforge\envs\mslice_new\lib\site-packages\matplotlib\colorbar.py", line 1144, in _process_values
    b = self.norm.inverse(b)
  File "C:\Users\dsc49661\AppData\Local\mambaforge\envs\mslice_new\lib\site-packages\matplotlib\colors.py", line 1554, in inverse
    raise ValueError("Invalid vmin or vmax")
ValueError: Invalid vmin or vmax
Traceback (most recent call last):
  File "C:\Users\dsc49661\AppData\Local\mambaforge\envs\mslice_new\lib\site-packages\matplotlib\cbook\__init__.py", line 287, in process
    func(*args, **kwargs)
  File "C:\Users\dsc49661\AppData\Local\mambaforge\envs\mslice_new\lib\site-packages\matplotlib\colorbar.py", line 551, in update_normal
    self.draw_all()
  File "C:\Users\dsc49661\AppData\Local\mambaforge\envs\mslice_new\lib\site-packages\matplotlib\colorbar.py", line 578, in draw_all
    self._process_values()
  File "C:\Users\dsc49661\AppData\Local\mambaforge\envs\mslice_new\lib\site-packages\matplotlib\colorbar.py", line 1144, in _process_values
    b = self.norm.inverse(b)
  File "C:\Users\dsc49661\AppData\Local\mambaforge\envs\mslice_new\lib\site-packages\matplotlib\colors.py", line 1554, in inverse
    raise ValueError("Invalid vmin or vmax")
ValueError: Invalid vmin or vmax
Traceback (most recent call last):
  File "C:\Users\dsc49661\AppData\Local\mambaforge\envs\mslice_new\lib\site-packages\mslice\presenters\quick_options_presenter.py", line 30, in <lambda>
    view.ok_clicked.connect(lambda: _set_axis_options(view, target, model, has_logarithmic, grid))
  File "C:\Users\dsc49661\AppData\Local\mambaforge\envs\mslice_new\lib\site-packages\mslice\presenters\quick_options_presenter.py", line 51, in _set_axis_options
    setattr(model, target[:-5] + 'log', view.log_scale.isChecked())
  File "C:\Users\dsc49661\AppData\Local\mambaforge\envs\mslice_new\lib\site-packages\mslice\plotting\plot_window\slice_plot.py", line 464, in colorbar_log
    self.change_axis_scale(self.colorbar_range, value)
  File "C:\Users\dsc49661\AppData\Local\mambaforge\envs\mslice_new\lib\site-packages\mslice\plotting\plot_window\slice_plot.py", line 216, in change_axis_scale
    self._canvas.figure.colorbar(colormesh)
  File "C:\Users\dsc49661\AppData\Local\mambaforge\envs\mslice_new\lib\site-packages\matplotlib\figure.py", line 1210, in colorbar
    cb = cbar.Colorbar(cax, mappable, **cb_kw)
  File "C:\Users\dsc49661\AppData\Local\mambaforge\envs\mslice_new\lib\site-packages\matplotlib\colorbar.py", line 483, in __init__
    self._reset_locator_formatter_scale()
  File "C:\Users\dsc49661\AppData\Local\mambaforge\envs\mslice_new\lib\site-packages\matplotlib\colorbar.py", line 1206, in _reset_locator_formatter_scale
    self._process_values()
  File "C:\Users\dsc49661\AppData\Local\mambaforge\envs\mslice_new\lib\site-packages\matplotlib\colorbar.py", line 1144, in _process_values
[... 65 lines cut ...]
[... cut ...]
  File "C:\Users\dsc49661\AppData\Local\mambaforge\envs\mslice_new\lib\site-packages\matplotlib\colorbar.py", line 1206, in _reset_locator_formatter_scale
    self._process_values()
  File "C:\Users\dsc49661\AppData\Local\mambaforge\envs\mslice_new\lib\site-packages\matplotlib\colorbar.py", line 1144, in _process_values
    b = self.norm.inverse(b)
  File "C:\Users\dsc49661\AppData\Local\mambaforge\envs\mslice_new\lib\site-packages\matplotlib\colors.py", line 1554, in inverse
    raise ValueError("Invalid vmin or vmax")
ValueError: Invalid vmin or vmax
```

</details>

closed 2022-05-16 by SilkeSchomann