mantidproject/mantid #33845: Manual Testing Project Recovery

Harrietbrown, 2022-05-06:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

http://developer.mantidproject.org/Testing/ErrorReporter-ProjectRecovery/ProjectRecoveryTesting.html

RichardWaiteSTFC, 2022-05-13:
3D plots (wireframe and surface) do not appear to be saved by project recovery

RichardWaiteSTFC, 2022-05-13:
Superplot isn't save dwith project recovery, but when you open workbench again and click Yes to recover it loads an old recovery file

RichardWaiteSTFC, 2022-05-13:
Project recovery doesn't seem to work with two mantid workbench windows open on IDAaaS (see section 5 of instructions) - in this case the project recovery window didn't pop up on opening. But my project recovery is behaving strangely since superplot...

RichardWaiteSTFC, 2022-05-13:
Project recovery is behaving strangely even with one instance of workbench open in IDAaaS - as before it doesn't have the recovery pop up for the first few goes and then sometimes it seems to load an older recovery file.

RichardWaiteSTFC, 2022-05-13:
Made an issue for IDAaaS #33903 - no other issues of note

closed 2022-05-13 by RichardWaiteSTFC