mantidproject/mantid #33846: Manual Testing Error Reporter

Harrietbrown, 2022-05-06:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

http://developer.mantidproject.org/Testing/ErrorReporter-ProjectRecovery/ErrorReporterTesting.html

DannyHindson, 2022-05-12:
Just two things:
- test 2. The private contacts details from test 1 were displayed on IDAaaS (behaves as instructions on Windows though)
- Also instructions suggest that pressing the "Don't share any information" button does actually submit an error report - see tests 1 and 3. Is that right? Don't think we saw anything coming through on the error-reports channel here

DannyHindson, 2022-05-12:
We did the testing based on looking for error reports appearing in the error-reports Slack channel rather than getting hold of someone with access to the error reports database. Seemed to work OK so the instructions could perhaps be updated to provide this an option

closed 2022-05-16 by Harrietbrown