mantidproject/mantid #33847: Manual Testing Sliceviewer

Harrietbrown, 2022-05-06:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

https://developer.mantidproject.org/Testing/SliceViewer/SliceViewer.html

RichardWaiteSTFC, 2022-05-10:
To see the latest cut viewer testing instructions go to this link
https://github.com/mantidproject/mantid/blob/main/dev-docs/source/Testing/SliceViewer/SliceViewer.rst

ConorMFinn, 2022-05-10:
@thomashampson @sf1919 While testing out the sliceviewer, we've noticed that the packaged versions on Babylon appear to be missing all their internal mantid help docs (on windows and mac at least). Is this intended or is it an issue with the build?

sf1919, 2022-05-11:
Thanks for flagging this @ConorMFinn . Not intended, definitely an issue with the package.

ConorMFinn, 2022-05-11:
Issues found as of 2022-05-10

- [ ] Unhandled exception when changing colourscale to log when empty bins are present and autoscaling is ticked (CNCS_7860_event.nxs)
- [ ] Grid lines are not shown when line plots are enabled
- [ ] Unhandled exception when the peak overlay is open and L & E are selected to be X and Y on a 4D workspace
- [ ] Rerunning the script while the 3D workspace is open and then panning the view around causes an unhandled exception
- [ ] Sliceviewer is not closed when md_4d is deleted from the ADS
- [ ] Dimension sliders jump around on macOS

RichardWaiteSTFC, 2022-05-12:
@DanielMurphy22 @Harrietbrown - I've created issues arising from this, should we close this manual testing issue or use it a bit like an epic?

DanielMurphy22, 2022-05-12:
> @DanielMurphy22 @Harrietbrown - I've created issues arising from this, should we close this manual testing issue or use it a bit like an epic?

I'd say once the testing is done and all the bugfix issues are raised then close the testing task. It'll be interesting to see, but presumably you can still update the tickboxes above when the issue is closed?

closed 2022-05-12 by RichardWaiteSTFC