mantidproject/mantid #33875: Font + font menu have changed on Linux Tar package on Ubuntu

DanielMurphy22, 2022-05-11:
When installing for manual testing, found on the mantidworkbenchnightly-6.3.20220510.2145.xz tar file for Linux

**Describe the bug**
The default font has changed. This could be intentional and may be required to support all flavours of Linux. The new font seems to be Sans Serif and the old one was Ubuntu Default. See the images below for the comeparison. It's a big change to the feel of the interface and we may need to actively choose a new font to support all Linux systems.

**Screenshots**
On an old nightly:
[image]

New tar installed version:
[image]

**Platform/Version (please complete the following information):**
 - OS: Ubuntu 18.04
 - Mantid Version 6.3.20220510.2145

DannyHindson, 2022-10-17:
Have just come across this font issue again while doing 6.5 smoke testing. I see what's described above but in addition I also get this error on startup of workbench:

[image]

and further when I change the font from Sans Serif to something else (eg the font called "Ubuntu" - I don't have "Ubuntu Default") then it seems like only some of the text gets updated to the new font. This is after having changed the font and restarted workbench:

[image]

So the font used for the menu has updated but some of the other text is stuck as Sans Serif

martyngigg, 2022-11-24:
I have just come across something similar on a completely unrelated project. When you get a second could you try running

```sh
export FONTCONFIG_PATH=/etc/fonts
mantidworkbench/bin/mantidworkbench
```
to see if that fixes the issue?

DannyHindson, 2022-11-25:
That didn't work for me. Although this did:
```
export FONTCONFIG_FILE=/etc/fonts/fonts.conf
```
Using that on my Ubuntu VM, I no longer get the Fontconfig error on start up and the fonts look correct

martyngigg, 2022-11-25:
> That didn't work for me. Although this did:
> 
> ```
> export FONTCONFIG_FILE=/etc/fonts/fonts.conf
> ```
> 
> Using that on my Ubuntu VM, I no longer get the Fontconfig error on start up and the fonts look correct

Interesting. The file is there but the environment variable is not set for some reason. At least that is something to go on.

MialLewis, 2023-03-02:
Just noting that we had a user complain about the new font.

martyngigg, 2023-03-02:
It looks like I'm still subscribed to this issue so I got the notification and I just thought I'd record here that for me on IDAaaS I seem to see better fonts. This is the Nightly on an ALF IDAaaS instance for me:

[image]

I don't recall doing anything special but maybe I have some config files or something in my home directory... 🤔

martyngigg, 2023-03-02:
I also don't get the `fontconfig` error that Danny noted above when starting with

```sh
/opt/mambaforge/bin/conda run -p /opt/mantidworkbenchnightly mantidworkbench
```

I hope this might be helpful!

jhaigh0, 2024-04-02:
Default font looks okay now. Will reopen if we get complaints again

closed 2024-04-02 by jhaigh0