mantidproject/mantid #33881: Docs Missing on New Packaged Nightlies

ConorMFinn, 2022-05-12:
Note: This and #33870 may be related

**Describe the bug**
On the new packaged versions (the ones on babylon) the opening the docs finder shows the finder but is completely blank, missing all docs. 

**To Reproduce**
1. Launch workbench
2. Open docs through Help -> Mantid Help
3. Open docs from any of the `?` buttons.

**Expected behavior**
The relevant documentation should be shown and it should be possible to navigate to all docs from the window that pops up. 

**Screenshots**
[image]

**Platform/Version (please complete the following information):**
 - OS: Windows, macOS, CentOS (Ubuntu has a crash: #33870)
 - Mantid Version: Packaged Nightlies

**Additional context**
From the debug log after opening:
```
Trying "/Applications/MantidWorkbenchNightly.app/Contents/Resources/bin/MantidProject.qhc"
QHelp Collection file /Applications/MantidWorkbenchNightly.app/Contents/Resources/bin/MantidProject.qhc not found
Trying "/Applications/MantidWorkbenchNightly.app/Contents/Resources/share/doc/MantidProject.qhc"
Using collection file "/Applications/MantidWorkbenchNightly.app/Contents/Resources/share/doc/MantidProject.qhc"
Failed to determine help cache file location
Removing help cache file "/Users/conor.finn/Library/Application Support/data/mantidproject/MantidProject.qhc"
Loading /Applications/MantidWorkbenchNightly.app/Contents/Resources/share/doc/MantidProject.qhc
Making local cache copy for saving information at /Users/conor.finn/Library/Application Support/data/mantidproject/MantidProject.qhc
helpengine.setupData() returned 1
open help window for "qthelp://org.mantidproject/doc/index.html"
```

SilkeSchomann, 2022-05-13:
The conda package for Windows crashes whenever the help menu is accessed, regardless whether it is via the menu or ? or from the about page.

martyngigg, 2022-05-16:
Fixed by #33922

PR #33922 (2022-05-16): Fix display of bundled documentation in workbench
**Note: Please merge #33888 first**

**Description of work.**

Adds fixes to the standalone packaging to fix the built-in documentation within workbench.

The fix is in the second commit and some cleanup and refactoring was done in the first.

**To test:**

This needs testing on all 3 platforms. You also need to have `mamba` available on your `PATH`. 

Checkout this pull request and

**Windows**:
- cd to `<source>/installers/conda/win`
- `./create_package -c mantid/label/nightly -s Unstable`
- a `.exe` will appear in the current working directo [... cut]
closed 2022-05-16 by martyngigg