mantidproject/mantid #33882: Workspace in sliceviewer not updated after in-place binary operation

RichardWaiteSTFC, 2022-05-12:


DanielMurphy22, 2022-05-19:
An additioanl bug is that if I run `md_3D = mtd['md_3D'] * 2` then the workspace has had the binary operation applied, but the sliceviewer replace handle hasn't been called (If I change X and Y axes, then the colorbar scale does indeed double!)

gemmaguest, 2022-05-25:
@DanielMurphy22 Looking at the stack trace in more detail from our debugging session earlier, I've spotted where the problem is.  We know the algorithm has run `lockWorkspaces` and has the write lock on the Poco::RWLock in the Workspace. Because we ran it in the interpreter it is running in Thread 1.

When the sliceviewer presenter `refresh_view` calls `readLock`, it ends up [here](https://github.com/pocoproject/poco/blob/master/Foundation/include/Poco/RWLock_POSIX.h#L51) on Linux. 

`pthread_rwlock_rdlock` fails to get the lock, hence Poco throws. Search for `EDEADLK` in [these docs](https://linux.die.net/man/3/pthread_rwlock_rdlock) - the reason is that this is also running in Thread 1, so it can't take the lock because we'd have a deadlock situation.

This explains why your fix to re-introduce the delayed refresh works - it will allow it to return to the event loop so the algorithm can finish processing and unlock before Sliceviewer refreshes. Whether that will still cause other problems, though, I'm not sure! I'll have a think about whether there might be another way around the problem.