mantidproject/mantid #33885: Mantid crashes if you do "Show Data" on a live data workspace

DannyHindson, 2022-05-12:
Crash on show data on "live" workspace in testing instructions section "Live Histogram Data". Instrument=ISIS_Histogram
Also crashes on show data on "live" workspace on "Live event pre and post processing". Instrument=ADARA_FakeEvent
Stack trace:

Name: Not provided Email: Not provided
Additional text:
Error from show data on live data (manual test)
Stack Trace:
Traceback (most recent call last):
File "/opt/mantidworkbenchnightly/lib/python3.8/site-packages/mantidqt/widgets/workspacedisplay/matrix/presenter.py", line 71, in action_replace_workspace
if self.model.workspace_equals(workspace_name):
File "/opt/mantidworkbenchnightly/lib/python3.8/site-packages/mantidqt/widgets/workspacedisplay/matrix/model.py", line 52, in workspace_equals
return workspace_name == self._ws.name()
RuntimeError: Variable invalidated, data has been deleted.

DanielMurphy22, 2022-05-13:
Not a problem on Release 6.3

DanielMurphy22, 2022-05-13:
If I comment out this Line 71 and always run the contents of the if statement then the behaviour seems to be fixed. I don't quite have the understanding yet so this is treating a symptom not the cause: https://github.com/mantidproject/mantid/blob/main/qt/python/mantidqt/mantidqt/widgets/workspacedisplay/matrix/presenter.py#L71

DanielMurphy22, 2022-05-13:
This is the first bad commit: https://github.com/mantidproject/mantid/commit/5a8567449036e7c9253be45821a246444e8d9946

DanielMurphy22, 2022-05-13:
This bug does not exist for ShowData on TableWorkspaces

DanielMurphy22, 2022-05-16:
Fixed by #33916

PR #33916 (2022-05-13): Save workspace name in workspacedisplay.model and allow renaming
**Description of work.**

Save the current workspace name in the Show Data (workspacedisplay) model. Since weak pointers were implemented (https://github.com/mantidproject/mantid/commit/5a8567449036e7c9253be45821a246444e8d9946) the workspace is invalidated by the time we were asking for the name. Now it is stored and updated when renamed.
**To test:**

- Load any workspace
- Right click -> show data
- Load the same workspace with the same name to overwrite the data

- Try the above but Load > Live Data
- Try some other things inc [... cut]
closed 2022-05-16 by DanielMurphy22