mantidproject/mantid #33886: Matplotlib 3.5 producing deprecation warnings

martyngigg, 2022-05-12:
**Describe the bug**

We swept up a lot of issues when we moved to matplotlib 3.5 but a few more places are showing deprecation warnings from 3.5. This is an umbrella issue to tackle those found in testing #33843.

**To Reproduce**

##### Colorfill plots

When testing the "Loading & Display Data" section of the [mantid basics course](http://docs.mantidproject.org/tutorials/mantid_basic_course/index.html#mantid-basic-course). Plotting the colorfill of MAR1060 gives 2 warnings:

```Scale is set to logarithmic so non-positive min value has been changed to 0.0001.
Support for case-insensitive scales is deprecated since 3.5 and support will be removed two minor releases later.
```
The first line is expected as the min scale value is too small.
The second deprecation warning has come from matplotlib 3.5 and should be suppressed/fixed.

##### Sliceviewer ROI

When testing the sliceviewer section of the basics course, loaded MAR11060, selected "Region selector" button and a warning:

```
The 'drawtype' parameter of __init__() was deprecated in Matplotlib 3.5 and will be removed two minor releases later. If any parameter follows 'drawtype', they should be passed as keyword, not positionally.
```

##### `get_window_title`

This was more difficult to reproduce but a deprecation warning popped up concerning `get_window_title`. A suggestion would be to search for uses of this in the code base.

**Expected behavior**

We should see only warnings from our code and not matplotlib 3.5 deprecation issues.

**Platform/Version (please complete the following information):**
All conda-based packages.

martyngigg, 2022-05-12:
I have marked this high priority as even though they are warnings they can cause confusion for users.

RichardWaiteSTFC, 2022-05-13:
I also see this one when I was fiddling around with errorbars
`The modification of the Axes.lines property was deprecated in Matplotlib 3.5 and will be removed two minor releases later. Use Axes.add_line instead.`

RichardWaiteSTFC, 2022-05-13:
Plotting a guess curve in fit browser zooms out as if home button clicked

mantid-builder, 2022-05-19:
Fixed by #33936

closed 2022-05-19 by mantid-builder

robertapplin, 2022-05-20:
Should this issue still be open @ConorMFinn , or has the last deprecation warning in the list been fixed? I've seen the same deprecation warning in this issue #33950

PR #33936 (2022-05-18): Fix Deprecation Warnings from matplotlib
**Description of work.**
Found and fixed some of the depreaction warnings caused by the matplotlib 3.5 upgrade. 

**To test:**
1. Plot something
2. Switch the scales to log (you may have to do this from the figure options (cog) if another issue hasn't been fixed yet)
3. Open the sliceviewer and draw a box on the plot (the button of a rectangle with squares in the corners)
4. Open the fit property browser

None of these things should come up with deprecation warnings. 

Fixes #33886 

*This does not require release notes* because **these issues have only [... cut]