mantidproject/mantid #33900: Settings cannot open if multiple mantids are opened in quick succession

thomashampson, 2022-05-13:
**Describe the bug**
After opening two instances of Mantid in quick succession (don't wait for the first to finish loading before opening the second), the settings dialog will not open. Instead the following exception occurs.

```
Traceback (most recent call last):
  File "C:\Users\dsc49661\AppData\Local\mambaforge\envs\mantid_manual_testing\lib\site-packages\workbench\app\mainwindow.py", line 684, in open_settings_window
    settings = SettingsPresenter(self)
  File "C:\Users\dsc49661\AppData\Local\mambaforge\envs\mantid_manual_testing\lib\site-packages\workbench\widgets\settings\presenter.py", line 37, in __init__
    self.general_settings = general_settings if general_settings else GeneralSettings(parent, None, self)
  File "C:\Users\dsc49661\AppData\Local\mambaforge\envs\mantid_manual_testing\lib\site-packages\workbench\widgets\settings\general\presenter.py", line 54, in __init__
    self.load_current_setting_values()
  File "C:\Users\dsc49661\AppData\Local\mambaforge\envs\mantid_manual_testing\lib\site-packages\workbench\widgets\settings\general\presenter.py", line 192, in load_current_setting_values
    self.view.completion_enabled.setChecked(completion_enabled)
TypeError: setChecked(self, bool): argument 1 has unexpected type 'str'

```

**To Reproduce**
1. Open Mantid twice in quick succession
2. Go to File > Settings in both Mantids
3. One will not work and throw up the error reporter

**Expected behavior**
Settings should open in both windows. This is fine if you wait for Mantid to open before starting the second instance.

**Screenshots**

**Platform/Version (please complete the following information):**
 - OS: all

gemmaguest, 2022-05-20:
Check if this is a regression - if not it's probably a bit of an edge case we're unlikely to hit

mantid-builder, 2022-06-08:
Fixed by #34020

PR #34020 (2022-06-08): Convert QSetting properties to boolean if a boolean is expected
**Description of work.**
When opening multiple Mantids in quick succession, there is sometimes a race condition between the syncing of the QSettings, and the 'fixup' of the booleans values such as changing 'true' to True. The syncing can be triggered by the event loop, and if this is triggered at the correct time, the converted booleans can be overwritten back to their original string form for a specific mantid instance. When this happens, there are errors.

The solution I have gone for is to use Pyside2 to specify the type of the [... cut]