mantidproject/mantid #33903: Project recovery on IDAaaS not using most recent recovery file

RichardWaiteSTFC, 2022-05-13:
**Describe the bug**
Project recovery on IDAaaS always seems to use the previous recovery file

**To Reproduce**
(1) Set project recovery to save every ~5s (File > Settings > General)
(2) Make a workspace
`
CreateWorkspace(DataX=range(12), DataY=range(12), DataE=range(12), NSpec=4, OutputWorkspace='NewWorkspace')
`
(3) Wait for it to save project (see this in log at debug level)
(4) Crash mantid with `Segfault `algorithm with `DryRun=False`
(5) Open workbench again (project recovery should give a pop-up askign to restore workspace but not on IDAaaS)
(6) Make 2 workspaces
```
CreateWorkspace(DataX=range(12), DataY=range(12), DataE=range(12), NSpec=4, OutputWorkspace='NewWorkspace')
CreateWorkspace(DataX=range(12), DataY=range(12), DataE=range(12), NSpec=4, OutputWorkspace='NewWorkspace2')
```
(7) Repeat steps (3-5) - this time I see a project recovery pop-up but it only restores one workspace

**Screenshots**

**Platform/Version (please complete the following information):**
 - Mantid nightly 9th May on IDAaaS

**Additional context**

AnthonyLim23, 2022-05-16:
Just done the first 3 steps on IDAaaS using the release and it recovered the workspace.

AnthonyLim23, 2022-05-16:
Just reproduced it on nightly. So looks like this a new issue

mantid-builder, 2022-06-01:
Fixed by #33988

closed 2022-06-01 by mantid-builder