mantidproject/mantid #33908: Investigate print button on a plot causing a freeze for 20 seconds

robertapplin, 2022-05-13:
**Description**
When clicking the print button on a plot, it seems to freeze for around 20 seconds, and sometimes says 'Not Responding', before finally opening a print dialog. I think it might be worth investigating why this is taking so long

**To Reproduce**
1. Load any data
2. Plot a spectrum
3. Click on the print button in the toolbar. It should take a while for the print dialog to load.

In my case it also said 'Not Responding':
[image]

robertapplin, 2022-05-13:
Adding milestone 6.4 for now so someone else can decide if this is important enough to investigate before the release

gemmaguest, 2022-05-20:
May have been around for a while. Investigate whether there is anything we can do here.

Harrietbrown, 2022-06-07:
Looking into this it looks like it is due to a delay in discoving networked printers. It does not look like something we can do much about so I'm closing this.

closed 2022-06-07 by Harrietbrown