mantidproject/mantid #33909: Zoom on guess curve updated fit browser stops sliders being useful

RichardWaiteSTFC, 2022-05-13:
**Describe the bug**
Plotting a guess curve in fit browser zooms out as if home button clicked - this is OK, although slightly unexpected. better if zoom level unchanged, or at least zoomed out enough to encompass the xrange of the fit.

However, when the guess is plotted and e.g. moving FWHM of a peak which then changes the position of the mouse and therefore the slider - making it difficult to use the slider

**To Reproduce**
(1) Make a workspace
```
ws = CreateSampleWorkspace(Function="User Defined", UserDefinedFunction="name=LinearBackground, A0=0.3;name=Gaussian, 
PeakCentre=5, Height=10, Sigma=0.7", NumBanks=1, BankPixelWidth=1, XMin=0, XMax=50, BinWidth=0.1, WorkspaceType="Event")
```
(2) Plot and zoom in on peak
(3) Open fit browser
(4) Add a peak
(5) Display > PLot guess (will zoom out)
(6) Zoom back in on peak
(7) Try moving FWHM slider (will zoom out)

**Expected behavior**
Zoom level unchanged

**Screenshots**

**Platform/Version (please complete the following information):**
 - IDAaaS mantid nightly 9th May

**Additional context**

DannyHindson, 2022-05-19:
The problem was introduced in this commit:

https://github.com/mantidproject/mantid/commit/053b57df23abdafe0fa56e6b6b54cdf886f10f0c - Remove autoscale_on_update option as it no longer functions

DannyHindson, 2022-05-19:
The original autoscale_on_update code was removed because some of the get_auto_scaleXXX functions called by it weren't present in matplotlib 3.5.

I think that's because there's a typo\naming problem in the decorator at the point it was removed. The functions get_autoscale_onx and get_autoscale_ony aren't in matplotlib but the functions get_autoscalex_on and get_autoscaley_on are (in both 3.1.2 and 3.5.0)

Think I'll reinstate the original decorator with this naming problem resolved.