mantidproject/mantid #33911: Crash installing script repo in IDAaaS

RichardWaiteSTFC, 2022-05-13:
**Describe the bug**
Installing script_repo on nightly in IDAaaS causes hard crash (with no stack trace) - but it does look like the script repo is installed (and it seems to function)!

**To Reproduce**
(1) File > Script Repository
(2) Select a folder
(3) Crash...

**Expected behavior**
No crash...

**Screenshots**

**Platform/Version (please complete the following information):**
Mantid nightly 9th May

**Additional context**

ConorMFinn, 2022-05-27:
I don't seem to be able to reproduce this on the latest nightly anymore (2022-05-25). Could you check if you can @RichardWaiteSTFC?

RichardWaiteSTFC, 2022-05-27:
> I don't seem to be able to reproduce this on the latest nightly anymore (2022-05-25). Could you check if you can @RichardWaiteSTFC?
Will try again on IDAaaS - in case it's useful you can 'uninstall' the script repo by just deleting the folder it points to!

RichardWaiteSTFC, 2022-05-27:
@ConorMFinn - I don't seem to be able to reproduce this on IDAaaS using same nightly (25/05) either, so am closing the issue!

closed 2022-05-27 by RichardWaiteSTFC