mantidproject/mantid #33915: Crash when selecting Plot Options after running script

SilkeSchomann, 2022-05-13:
**Describe the bug**
When running the example script from the tutorial (https://docs.mantidproject.org/nightly/tutorials/python_in_mantid/plotting/02_scripting_plots.html#example-script) and clicking on Plot Options, Mantid crashes reproducable.

**Screenshots**
[image]

robertapplin, 2022-05-17:
A similar related error is seen when using a matplotlib version >= 3.2.0 (i.e. when using a conda build):
[image]

robertapplin, 2022-05-20:
Fixed by #33815

closed 2022-05-20 by robertapplin

mantid-builder, 2022-05-20:
Fixed by #33935