mantidproject/mantid #33937: Monments still giving NAN's

AnthonyLim23, 2022-05-17:
The moments tab of indirect's reduction GUI is giving NAN's for everything except the first moment. It looks like this is caused by https://github.com/mantidproject/mantid/blob/main/Framework/PythonInterface/plugins/algorithms/WorkflowAlgorithms/SofQWMoments.py#L141

**Describe the bug**

1. Open Indirect->reduction
2. In the energy transfer tab use run `26176`
3. Press run
4. Go to the S(Q, w) tab
5. Change from file to workspace and select the one you just made
6. Tick the rebin in energy option
7. Set E Low to -0.5
8. Set E High to 0.5
9. Press run
10. Go to the moments tab
11. Change from file to workspace and select the one you just made
12. Press run -> a single line will appear. Inspecting the data from the ADS shows that the other spectra are all NAN's

**Expected behavior**

**Screenshots**

**Platform/Version (please complete the following information):**
 - OS: [e.g. Windows, RHEL 7, Ubuntu, macOS]
 - OS Version:
 - Mantid Version [e.g. 6.0.0]

**Additional context**

mducle, 2022-05-23:
Hi @AnthonyLim23 @Harrietbrown sorry about this... I've created a PR which I think fixes things now... I'm getting exactly the same result as with Mantid 6.2... To be honest I'm not 100% sure that this result is _correct_ - maybe you can ask Spencer to look at the algorithms and test the data with different bin sizes...? Anyway, at least it's consistent with previous behaviour!

I've changed it so that when `ConvertToPointData` is called on a `RebinnedOutput` created by `SofQW3` it explicitly sets the fractions to unity (previously it implicitly set them to zero, hence the `NaN`s). This effectively makes the new behaviour of `Integration` the same as the old behaviour (where it ignored the fractions but _implicitly_ treats them as unity; now it is aware of the fractions and _sees_ that they are unity because they have been set that way by `ConvertToPointData`.)

This behaviour seems the most intuitive to me since conversion to points means you discard the binning information and hence the bin fractions don't make sense (plus it also works, unlike the case where the fractions are preserved which produces non-sensical moments [but I'm not sure why]).

I need to update the documentation for `ConvertToPointData` and `ConvertToHistogram` as well as do the release docs...

mantid-builder, 2022-06-07:
Fixed by #33972

closed 2022-06-07 by mantid-builder