mantidproject/mantid #33952: Slicepoint spin box not updated when select peak outside data extent in sliceviewer

RichardWaiteSTFC, 2022-05-19:
**Describe the bug**
When the user selects a peak in the peak viewer table in sliceviewer the colorfill plot updates to show the region of the peak - at the slicepoint corresponding to the peak position along the integrated dimension. However, if a peak with a position outside the extent of the data along the integrated dimension, the slicepoint in the view (the spin-boxes of the dimension widget) is not updated. Though the call to BinMD uses the correct slicepoint of the peak.

**To Reproduce**
(1) Make some data

```
ws_4D = CreateMDWorkspace(Dimensions=4, Extents=[-1, 1, -1, 1, -1, 1, -1, 1], Names="H,K,L,E",
                              Frames='HKL,HKL,HKL,General Frame', Units='r.l.u.,r.l.u.,r.l.u.,meV')
FakeMDEventData(ws_4D, UniformParams=1e6)
expt_info_4D = CreateSampleWorkspace()
ws_4D.addExperimentInfo(expt_info_4D)
SetUB(ws_4D, 1, 1, 2, 90, 90, 120)
CreatePeaksWorkspace(InstrumentWorkspace='ws_4D', NumberOfPeaks=0, OutputWorkspace='peaks')
CopySample(InputWorkspace='ws_4D', OutputWorkspace='peaks', CopyName=False, CopyMaterial=False, CopyEnvironment=False, CopyShape=False)
AddPeakHKL(Workspace='peaks', HKL='1,2,3')

```

(2) Open `ws_4D` in sliceviewer
(3) Overlay `peaks`
(4) Select the peak row in the table 
- the plot should go green (previously showed random fake data)
- the slicepoint in the spin box in the dimensions above the colorfill still reads -0.99
- the history of `ws_4D_svrebinned` shows that `BinMD` was called with extents [2.95,3.05] along L
(5) In the peak actions above the table click `Add Peaks`
(6) Click on the colorfill plot 
- This should add a peak, the slicepoint will be the same as the peak row above (i.e. 3)

**Expected behavior**
The slicepoint in the spinbox in the view would read 3 (the slicepoint of the peak selected) - it looks like the spin box just isn't being updated.

**Screenshots**

**Platform/Version (please complete the following information):**
 - Seen in mantid 6.2 and mantid 6.3 nightly from 19th May

**Additional context**

jclarkeSTFC, 2023-11-23:
Hi @RichardWaiteSTFC, this looks like the script is creating a peak outside of the data range, e.g. the peak is at L=3, but the range of the data is [-1,1], is this a thing that happens normally? Should `AddPeakHKL` be complaining?

RichardWaiteSTFC, 2023-11-23:
> Hi @RichardWaiteSTFC, this looks like the script is creating a peak outside of the data range, e.g. the peak is at L=3, but the range of the data is [-1,1], is this a thing that happens normally? Should `AddPeakHKL` be complaining?

It can happen if the peaks are created using `PredictPeak` or similar. From what I remember the data are rebinned correctly (all 0s if outside the range) but the slicepoint spin box seems to get out of sync

PR #36473 (2023-12-04): Fix Slice Viewer spin box for peaks when outside of data range
Overlaying a peak in Slice Viewer will now always result in the corresponding spin boxes at the top of the interface showing the value of that dimension that corresponds to the peak, even if the peak is outside the range of the data. Previously these boxed were clipped to the min/max of the data.

#### Summary of work

- Extended range of spin box to cover peak when necessary

Fixes #33952. 

### To test:
See #33952 for original issue. Other peak overlaying and dimension-changing behaviour in Slice Viewer should be unaffected.

--- [... cut]
closed 2023-12-04 by SilkeSchomann