mantidproject/mantid #33965: SANS Interface: Unable to add runs containing multiple periods

smk78, 2022-05-21:
Reported by Dirk (@dehoni):

**Describe the bug**
The SANS GUI does not properly add runs that contain multiple periods. No error is produced. An added file is created, together with an "add_monitor.nxs" not present for single period runs. Trying to use the file created in the reduction table turns the line blue with the tag "Ended up in added branch where it is neither processed or raw?". Normal measurements with 1 period and polarised raw files as they come get reduced as expected.

**To Reproduce**
Please use any two files from the LARMOR experiments RB2210159 or  2210170.

    Open the ISIS SANS interface
    Click Sum runs and add files
   In the reduction table call the added file and click process. The reduction will stall with the line turning blue.

**Expected behavior**
Mantid should show a more meaningful error message explaining that the file could not be found/is corrupted. 

Added nxs files containing periods should be processed with the SANS GUI as unadded nxs files with periods are.

**Platform/Version (please complete the following information):**
    OS: Idaas
    OS Version:
    Mantid Version 6.3.0

**Additional context**
Whilst this has been labelled for the attention of the ISIS LSS/SANS team, this may be a wider issue with the sum run functionality in Mantid for files with multiple periods.

smk78, 2022-05-25:
> @smk78 The high priority label means this is severe enough to delay the current release. Is that the case?

Check with Dirk how much of a barrier this is to what they are doing. But I suspect it's a big barrier.

And is this SANS-specific, or does it affect any instrument trying to add event data with periods?

gemmaguest, 2022-06-14:
Fixed by #34016

PR #34016 (2022-06-13): Fix a bug saving add files for multi period data
**Report to:** @smk78 @dehoni

This PR fixes a bug in the SANS GUI where summed files for multi-period data were not being saved correctly.
- There was a bug setting the filename for subsequent periods after the first one, where the save path was being pre-pended too many times. This meant that only the first period was saved; monitors were also not being saved.

Fixes #33965

**Description of work.**

**To test:**

- Find a LARMOR user file - here is a minimal version which you can save as e.g. `MaskFile.toml`. 
```
toml_file_version = 1

[inst [... cut]