mantidproject/mantid #33978: Move to numpy v1.22.4

sf1919, 2022-05-25:
**Describe the outcome that is desired.**
That Mantid Workbench can use v1.22.4 of numpy.

**Describe any solutions you are considering**
The new version of numpy causes problems with BayesQuasi and Elemental Analysis. The solution is to update these as necessary to work with newer versions of numpy.  

**Additional context**
Recently a new version of numpy was released (v1.22.4). Unfortunately this was not compatible with some of code base, specifically BayesQuasi and Elemental Analysis. We think it may have been caused by this change to numpy - https://github.com/numpy/numpy/pull/21154/files . 

Once this issue has been fixed the changes made in #33974 can be reverted.

The following is the stack trace for the 3 tests that failed when numpy 1.22.4 was used

Test 1 - BayesQuasiTest.BayesQuasiTest
```
Python-[Warning] Failed to load plugin /jenkins_workdir/workspace/pull_requests-conda-linux/Framework/PythonInterface/plugins/algorithms/WorkflowAlgorithms/BayesQuasi.py.
Error: Traceback (most recent call last):
  File "/jenkins_workdir/workspace/pull_requests-conda-linux/Framework/PythonInterface/mantid/kernel/plugins.py", line 184, in load_from_file
    name, module = load_plugin(filepath)
  File "/jenkins_workdir/workspace/pull_requests-conda-linux/Framework/PythonInterface/mantid/kernel/plugins.py", line 204, in load_plugin
    module = loader.run()
  File "/jenkins_workdir/workspace/pull_requests-conda-linux/Framework/PythonInterface/mantid/kernel/plugins.py", line 49, in run
    return SourceFileLoader(name, pathname).load_module()
  File "<frozen importlib._bootstrap_external>", line 522, in _check_name_wrapper
  File "<frozen importlib._bootstrap_external>", line 1022, in load_module
  File "<frozen importlib._bootstrap_external>", line 847, in load_module
  File "<frozen importlib._bootstrap>", line 265, in _load_module_shim
  File "<frozen importlib._bootstrap>", line 702, in _load
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 843, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/jenkins_workdir/workspace/pull_requests-conda-linux/Framework/PythonInterface/plugins/algorithms/WorkflowAlgorithms/BayesQuasi.py", line 12, in <module>
    from IndirectImport import *
  File "/jenkins_workdir/workspace/pull_requests-conda-linux/scripts/Inelastic/IndirectImport.py", line 14, in <module>
    import numpy.core.setup_common as numpy_cfg
  File "/jenkins_workdir/workspace/pull_requests-conda-linux/mambaforge/envs/mantid-developer/lib/python3.8/site-packages/numpy/core/setup_common.py", line 123, in <module>
    with open(pathlib.Path(__file__).parent / file) as f:
FileNotFoundError: [Errno 2] No such file or directory: '/jenkins_workdir/workspace/pull_requests-conda-linux/mambaforge/envs/mantid-developer/lib/python3.8/site-packages/numpy/core/feature_detection_locale.h'
```

Test 2 - Test 1 - BayesQuasiTest.BayesStretchTest
```
Python-[Warning] Failed to load plugin /jenkins_workdir/workspace/pull_requests-conda-linux/Framework/PythonInterface/plugins/algorithms/WorkflowAlgorithms/BayesStretch.py.
Error: Traceback (most recent call last):
  File "/jenkins_workdir/workspace/pull_requests-conda-linux/Framework/PythonInterface/mantid/kernel/plugins.py", line 184, in load_from_file
    name, module = load_plugin(filepath)
  File "/jenkins_workdir/workspace/pull_requests-conda-linux/Framework/PythonInterface/mantid/kernel/plugins.py", line 204, in load_plugin
    module = loader.run()
  File "/jenkins_workdir/workspace/pull_requests-conda-linux/Framework/PythonInterface/mantid/kernel/plugins.py", line 49, in run
    return SourceFileLoader(name, pathname).load_module()
  File "<frozen importlib._bootstrap_external>", line 522, in _check_name_wrapper
  File "<frozen importlib._bootstrap_external>", line 1022, in load_module
  File "<frozen importlib._bootstrap_external>", line 847, in load_module
  File "<frozen importlib._bootstrap>", line 265, in _load_module_shim
  File "<frozen importlib._bootstrap>", line 702, in _load
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 843, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/jenkins_workdir/workspace/pull_requests-conda-linux/Framework/PythonInterface/plugins/algorithms/WorkflowAlgorithms/BayesStretch.py", line 8, in <module>
    from IndirectImport import *
  File "/jenkins_workdir/workspace/pull_requests-conda-linux/scripts/Inelastic/IndirectImport.py", line 14, in <module>
    import numpy.core.setup_common as numpy_cfg
  File "/jenkins_workdir/workspace/pull_requests-conda-linux/mambaforge/envs/mantid-developer/lib/python3.8/site-packages/numpy/core/setup_common.py", line 123, in <module>
    with open(pathlib.Path(__file__).parent / file) as f:
FileNotFoundError: [Errno 2] No such file or directory: '/jenkins_workdir/workspace/pull_requests-conda-linux/mambaforge/envs/mantid-developer/lib/python3.8/site-packages/numpy/core/feature_detection_locale.h'
```

Test 3 - elemental_analysis_context_test
```
LoadInstrument-[Notice] LoadInstrument started
[... 42 lines cut ...]
======================================================================
ERROR: test_rebin (elemental_analysis_context_test.ElementalAnalysisContextTest)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/jenkins_workdir/workspace/pull_requests-conda-linux/Framework/PythonInterface/mantid/simpleapi.py", line 923, in do_set_property
    alg_object.setProperty(key, new_value)
ValueError: Invalid value for property InputWorkspace (Workspace) from string "9999; Detector 1": Workspace "9999; Detector 1" was not found in the Analysis Data Service

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/jenkins_workdir/workspace/pull_requests-conda-linux/mambaforge/envs/mantid-developer/lib/python3.8/unittest/mock.py", line 1325, in patched
    return func(*newargs, **newkeywargs)
  File "/jenkins_workdir/workspace/pull_requests-conda-linux/qt/python/mantidqtinterfaces/test/Muon/elemental_analysis_2/elemental_analysis_context_test.py", line 58, in test_rebin
    self.context._run_rebin("9999; Detector 1", "Variable", mock_params)
  File "/jenkins_workdir/workspace/pull_requests-conda-linux/qt/python/mantidqtinterfaces/mantidqtinterfaces/Muon/GUI/ElementalAnalysis2/context/context.py", line 117, in _run_rebin
    CloneWorkspace(InputWorkspace=raw_workspace, OutputWorkspace=rebined_run_name)
  File "/jenkins_workdir/workspace/pull_requests-conda-linux/Framework/PythonInterface/mantid/simpleapi.py", line 1056, in __call__
    set_properties(algm, *args, **final_keywords)
  File "/jenkins_workdir/workspace/pull_requests-conda-linux/Framework/PythonInterface/mantid/simpleapi.py", line 944, in set_properties
    do_set_property(key, value)
  File "/jenkins_workdir/workspace/pull_requests-conda-linux/Framework/PythonInterface/mantid/simpleapi.py", line 927, in do_set_property
    raise e.__class__(msg) from e
ValueError: Problem setting "InputWorkspace" in CloneWorkspace-v1: Invalid value for property InputWorkspace (Workspace) from string "9999; Detector 1": Workspace "9999; Detector 1" was not found in the Analysis Data Service
```

AnthonyLim23, 2022-05-25:
the bayesQuasifailures seem to come from checking the numpy version for the fortran code https://github.com/mantidproject/mantid/blob/main/scripts/Inelastic/IndirectImport.py#L100

mantid-builder, 2022-05-31:
Fixed by #33996

sf1919, 2022-06-01:
Unfortunately we are still having issues with elemental analysis test so this needs to be reopened whilst we investigate.

Recent stack trace

```01:38:33  1819/2580 Test #2137: python.MuonQt5.elemental_analysis_context_test.elemental_analysis_context_test ...............................................................***Failed   13.83 sec
01:38:33  FrameworkManager-[Notice] Welcome to Mantid 6.3.20220531.1603
01:38:33  FrameworkManager-[Notice] Please cite: http://dx.doi.org/10.1016/j.nima.2014.07.029 and this release: http://dx.doi.org/10.5286/Software/Mantid
01:38:33  LoadInstrument-[Notice] LoadInstrument started
01:38:33  LoadInstrument-[Notice] LoadInstrument successful, Duration 0.02 seconds
01:38:33  ..CreateSampleWorkspace-[Notice] CreateSampleWorkspace started
01:38:33  CreateSampleWorkspace-[Notice] CreateSampleWorkspace successful, Duration 0.29 seconds
01:38:33  CreateSampleWorkspace-[Notice] CreateSampleWorkspace started
01:38:33  CreateSampleWorkspace-[Notice] CreateSampleWorkspace successful, Duration 0.02 seconds
01:38:33  .CreateSampleWorkspace-[Notice] CreateSampleWorkspace started
01:38:33  CreateSampleWorkspace-[Notice] CreateSampleWorkspace successful, Duration 0.02 seconds
01:38:33  CreateSampleWorkspace-[Notice] CreateSampleWorkspace started
01:38:33  CreateSampleWorkspace-[Notice] CreateSampleWorkspace successful, Duration 0.03 seconds
01:38:33  .CreateSampleWorkspace-[Notice] CreateSampleWorkspace started
01:38:33  CreateSampleWorkspace-[Notice] CreateSampleWorkspace successful, Duration 0.03 seconds
01:38:33  CreateSampleWorkspace-[Notice] CreateSampleWorkspace started
01:38:33  CreateSampleWorkspace-[Notice] CreateSampleWorkspace successful, Duration 0.02 seconds
01:38:33  .CreateSampleWorkspace-[Notice] CreateSampleWorkspace started
01:38:33  CreateSampleWorkspace-[Notice] CreateSampleWorkspace successful, Duration 0.01 seconds
01:38:33  CreateSampleWorkspace-[Notice] CreateSampleWorkspace started
01:38:33  CreateSampleWorkspace-[Notice] CreateSampleWorkspace successful, Duration 0.01 seconds
01:38:33  .....CreateSampleWorkspace-[Notice] CreateSampleWorkspace started
01:38:33  CreateSampleWorkspace-[Notice] CreateSampleWorkspace successful, Duration 0.01 seconds
01:38:33  CreateSampleWorkspace-[Notice] CreateSampleWorkspace started
01:38:33  CreateSampleWorkspace-[Notice] CreateSampleWorkspace successful, Duration 0.01 seconds
01:38:33  ...CreateWorkspace-[Notice] CreateWorkspace started
01:38:33  CreateWorkspace-[Notice] CreateWorkspace successful, Duration 0.00 seconds
01:38:33  E...CreateWorkspace-[Notice] CreateWorkspace started
01:38:33  CreateWorkspace-[Notice] CreateWorkspace successful, Duration 0.00 seconds
01:38:33  ....
01:38:33  ======================================================================
01:38:33  ERROR: test_rebin (elemental_analysis_context_test.ElementalAnalysisContextTest)
01:38:33  ----------------------------------------------------------------------
01:38:33  Traceback (most recent call last):
01:38:33    File "c:\jenkins\workspace\main_nightly_deployment_prototype\framework\pythoninterface\mantid\simpleapi.py", line 923, in do_set_property
01:38:33      alg_object.setProperty(key, new_value)
01:38:33  ValueError: Invalid value for property InputWorkspace (Workspace) from string "9999; Detector 1": Workspace "9999; Detector 1" was not found in the Analysis Data Service
01:38:33  
01:38:33  The above exception was the direct cause of the following exception:
01:38:33  
01:38:33  Traceback (most recent call last):
01:38:33    File "C:\Jenkins\workspace\main_nightly_deployment_prototype\mambaforge\envs\mantid-developer\lib\unittest\mock.py", line 1325, in patched
01:38:33      return func(*newargs, **newkeywargs)
01:38:33    File "C:/Jenkins/workspace/main_nightly_deployment_prototype/qt/python/mantidqtinterfaces/test/Muon/elemental_analysis_2/elemental_analysis_context_test.py", line 58, in test_rebin
01:38:33      self.context._run_rebin("9999; Detector 1", "Variable", mock_params)
01:38:33    File "c:\jenkins\workspace\main_nightly_deployment_prototype\qt\python\mantidqtinterfaces\mantidqtinterfaces\Muon\GUI\ElementalAnalysis2\context\context.py", line 117, in _run_rebin
01:38:33      CloneWorkspace(InputWorkspace=raw_workspace, OutputWorkspace=rebined_run_name)
01:38:33    File "c:\jenkins\workspace\main_nightly_deployment_prototype\framework\pythoninterface\mantid\simpleapi.py", line 1056, in __call__
01:38:33      set_properties(algm, *args, **final_keywords)
01:38:33    File "c:\jenkins\workspace\main_nightly_deployment_prototype\framework\pythoninterface\mantid\simpleapi.py", line 944, in set_properties
01:38:33      do_set_property(key, value)
01:38:33    File "c:\jenkins\workspace\main_nightly_deployment_prototype\framework\pythoninterface\mantid\simpleapi.py", line 927, in do_set_property
01:38:33      raise e.__class__(msg) from e
01:38:33  ValueError: Problem setting "InputWorkspace" in CloneWorkspace-v1: Invalid value for property InputWorkspace (Workspace) from string "9999; Detector 1": Workspace "9999; Detector 1" was not found in the Analysis Data Service```

martyngigg, 2022-06-01:
Oh, it only fails on Windows...

mantid-builder, 2022-06-01:
Fixed by #33996

closed 2022-06-01 by mantid-builder