mantidproject/mantid #33980: Setting both axes to log on a Waterfall plot only affects first spectrum

DannyHindson, 2022-05-25:
**Original reporter:** Leide P. Cavalcanti, SANS

**Describe the bug**
-Water Fall graph does not accept logXlog mode for all curves in the same graph, just for the first one (please see attachment).

[Mantid_graphs.pdf](https://github.com/mantidproject/mantid/files/8769322/Mantid_graphs.pdf)

DH: I think this means both axes are set to log scale. I can reproduce this. The problem existed in 6.3 as well as the latest code so am labelling it as a 6.5 issue rather than an urgent issue to fix now

**To Reproduce**

**Expected behavior**

**Screenshots**

**Platform/Version (please complete the following information):**
 - OS: [e.g. Windows, RHEL 7, Ubuntu, macOS]
 - OS Version:
 - Mantid Version [e.g. 6.0.0]

**Additional context**

Harrietbrown, 2022-05-25:
I was just checking this for myself The waterfall plots do convert the scale to Log but the scales use the coordinates from the waterfall. that is to say if the workspaces has a first point at 10, but the waterfall plot translates that point to being at 20 the position when it is converted to log will be as if the value was 20 instead of log(10) + offset

[image]

[image]

DannyHindson, 2022-05-25:
Please can you send me the script you used to generate those plots and I'll have a look? If it's actually fine then I'll get back to the person who reported it

Harrietbrown, 2022-05-25:
```# import mantid algorithms, numpy and matplotlib
from mantid.simpleapi import *
import matplotlib.pyplot as plt
import numpy as np

x = np.arange(1,100)
y1 = 2*x+2
y2 = 3*x+3
y3 = 4*x+4
y4 = 5*x+5
y5 = 6*x+6
y0 = np.append(y1,y2)
y0 = np.append(y0,y3)
y0 = np.append(y0,y4)
y0 = np.append(y0,y5)
ws0 = CreateWorkspace(DataX=x, DataY=y0,NSpec=5)```

DannyHindson, 2022-05-26:
Just had a closer look at the script you sent through. I agree the curves are all being plotted with a log scale so the original description on the issue isn't quite right. However I think the fact that the offsets are calculated in a linear way regardless of whether the axes are set to log or not isn't very helpful.

With log scales on both axes, the offset in "canvas distance" between the first and second curves is quite large even for a small % offset (eg default of 10% for x offset). Also the offset in canvas distance isn't the same for different x values so the shapes of the curves relative to each other get distorted. This is what I get for example if I do a waterfall plot of 5 spectra that are the same but with log scales enabled:
[image]

I think it would make more sense for the offset calculation to switch to a log calculation if the axes are set to log. This would mean the offset is constant in terms of "canvas distance" regardless of whether the axes are log or not. I can go back to the user to check this but I think that's probably where they're coming from. What do you think?

Harrietbrown, 2022-05-26:
That sounds sencible. I'll look into a solution for this

Harrietbrown, 2022-05-26:
The issue comes with how can we have an offset on log scale. We would need to offset every point by a different amount, this doesn't look like a fard thing to do. I just need to work out the correct amount to offset since the offset is now dependant on the origional value

PR #34013 (2022-06-28): Fix waterfall for log axes in plots
**Description of work.**

This PR changes the offset for waterfall plots when using log scales so that they are equally distrebuted in log space

**To test:**

Run the following script to make a test workspace.
```
# import mantid algorithms, numpy and matplotlib
from mantid.simpleapi import *
import matplotlib.pyplot as plt
import numpy as np

x = np.arange(1,101)
e = 2*np.ones(500)
y1 = 2*x+2
y2 = 3*x+3
y3 = 4*x+4
y4 = 5*x+5
y5 = 6*x+6
y0 = np.append(y1,y2)
y0 = np.append(y0,y3)
y0 = np.append(y0,y4)
y0 = np.append(y0,y5)
ws0 = CreateWorkspace(DataX=x, Dat [... cut]
closed 2022-06-28 by gemmaguest