mantidproject/mantid #34000: Clearing time slicing settings has no effect in SANS GUI

gemmaguest, 2022-05-30:
**Original reporter:** Najet at ISIS

**Describe the bug**
If you enter time slicing settings in the SANS GUI and then clear them, when you process runs it continues to do time slicing, even though it should not.

**To Reproduce**
- Open the ISIS SANS GUI
- Load the user file `MaskFile.toml` from the TrainingCourseData (or use `MaskFile.txt` if using v6.3.0 or earlier)
- Load the batch file `batch_mode_reduction.csv` 
- On the Settings tab, under Slices, enter some range(s) e.g. `5-8`
- On the Runs tab, select the first row and click Process Selected. 
- When the reduction is finished there will be two workspaces with a suffix `t5.00_T8.00`
- Delete those workspaces.
- Clear the Slices text you entered and re-process the row.
- You should now get two workspaces without the suffix, but the suffix still appears, indicating time slicing is still being done.

**Expected behavior**
Time slicing should not be performed if the Slices settings box is empty.

**Platform/Version (please complete the following information):**
 - OS: All
 - OS Version:
 - Mantid Version: I tested this back to v6.0.0 and it was reproducible there as well as in the nightly  6.3.20220527.1456

**Additional Context**
Re-loading the user file will make it perform the correct reduction.

gemmaguest, 2022-11-10:
We probably need to rethink how the view/model interactions work. Possibly worth considering alongside #30306

ConorMFinn, 2023-05-18:
We still need to fix this.

cailafinn, 2023-11-15:
Still needs fixing.

closed 2025-08-21 by github-actions[bot]