mantidproject/mantid #34049: The X Axis Title of the graphs hides when we change size

DannyHindson, 2022-06-13:
This was raised as a feature request in a support email from Leide Cavalcanti that was sent on 18th May 2022 along with some beta testing feedback on the 6.4 release candidate.

The issue is #2 in the attached document

[Mantid_graphs.pdf](https://github.com/mantidproject/mantid/files/8891535/Mantid_graphs.pdf)

martyngigg, 2023-01-18:
I'm reopening this as it caused several issues uncovered by manual testing: https://github.com/mantidproject/mantid/issues/34973 https://github.com/mantidproject/mantid/issues/34974

mantid-builder, 2023-01-18:
Fixed by #34907

closed 2023-01-18 by mantid-builder

martyngigg, 2023-01-19:
The script we have to close issues when PRs are merged to `release-next` is going to keep catching this so I'll open a new one. We could edit the PR to remove `Fixes #` but that doesn't seem right either.

PR #34907 (2023-01-13): Stop axes labels being hidden by the plot window
**Description of work.**

This PR fixes the bug where the axes labels of a plot could be hidden outside the edge of the plot window after resizing.

**To test:**

- Load some data
- Plot a spectrum
- Play around with resizing the window and check that the axes labels remain visible up until matplotlib is forced to push them outside the window (almost minimum window size).
  - When this happens there should be one warning issued for horizontal and vertical limits respectively

It may be worth comparing to another branch to observe how it was prob [... cut]