mantidproject/mantid #34057: Windows Smoke Tests: Release 6.4

DanielMurphy22, 2022-06-14:
**Before testing**:
- Check this main issue relates to the OS you will test on.
- Tick the box beside a testing issue listed below to mark *you are testing this issue*
- Enter the testing task issue and assign yourself as for a normal Github issue.

**Afterwards**:
- Comment on the testing issue with any issues you came across.
- If no major issues were found, please **close the testing issue**.

If you have any questions please contact the creator of this issue.

### Tick below before testing the related issue

- [ ] #34051

- [ ] #34052

- [ ] #34053

- [ ] #34054

- [ ] #34055

- [ ] #34056

DannyHindson, 2022-06-14:
Not sure this fits into any of the sub issues but the plots aren't being generated in any of the algorithm doc pages eg this is from CalculatePlaczek. Same problem in DiscusMultipleScatteringCorrection:

[image]

DanielMurphy22, 2022-06-14:
closing as all issues seem to be raised

closed 2022-06-14 by DanielMurphy22