mantidproject/mantid #34069: Redhat Python Smoke Tests  :snake:

DanielMurphy22, 2022-06-14:
## If at ISIS, please test on IDAaaS
* Follow the instructions [here](https://stfc365.sharepoint.com/:w:/s/isiscomputingdivision-staffsite/ESTnjvxoislDsmpI8vxNyLUBZxw_51y-Qw0OklU_vNgXVA?e=Ih3iLK) to test Mantid on IDAaaS

**Before testing**:
- Check this testing issue relates to the OS you will test on.
- If unassigned, please assign yourself as for a normal Github issue.
- Please run these tests on the release package of Mantid; **not a locally built version**.

**Afterwards**:
- Comment below with any issues you came across.
- If no major issues were found, please **close this testing issue**.
- Check the main issue for this OS for other unassigned smoke tests.

If you have any questions please contact the creator of this issue.

 :snake:

* Test that the Python scripting window works, [directions here](https://docs.mantidproject.org/nightly/workbench/scriptwindow.html)
 - [ ] Editor options 
 - [ ] Execution options 
 - [ ] Script output  
- [ ] Perform some workspace algebra 
- [ ] Test numpy functionality 
- [ ] Use the scripting window to run some scripts 
- [ ] Run through some examples from [the documentation](https://docs.mantidproject.org/nightly/tutorials/python_in_mantid/index.html), 3 or 4 examples from the Solutions is enough

robertapplin, 2022-06-14:
Opened the `Fit Script Generator` to generate a fitting script, and clicked on the `?` button to get the documentation. It couldn't find anything:
```
Missing page - qthelp://org.mantidproject/doc/interfaces/utility/Fit Script Generator.html
```

robertapplin, 2022-06-14:
Ran the following script. When it opened the plot, I clicked the `?` button and there was a crash. This doesn't seem to happen all the time. I just kept closing and opening the documentation until I could reproduce it for a second time
```
# import mantid algorithms, numpy and matplotlib
from mantid.simpleapi import *
import matplotlib.pyplot as plt
import numpy as np

LoadMuonNexus(Filename=r'\\isis.cclrc.ac.uk\inst$\ndxmusr\instrument\data\cycle_16_5\MUSR00062260.nxs', OutputWorkspace='MUSR00062260.nxs', DeadTimeTable='MUSR00062260.nxs_deadtime_table', DetectorGroupingTable='__notUsed')
RenameWorkspace(InputWorkspace='MUSR00062260.nxs_deadtime_table', OutputWorkspace='MUSR62260_deadtime MA')
RenameWorkspace(InputWorkspace='MUSR00062260.nxs', OutputWorkspace='MUSR62260_raw_data MA')
GroupWorkspaces(InputWorkspaces='MUSR62260_raw_data MA,MUSR62260_deadtime MA', OutputWorkspace='MUSR62260 MA')
MuonPreProcess(InputWorkspace='MUSR62260_raw_data MA', OutputWorkspace='__MUSR62260_pre_processed_data', TimeMin=0.10199999999999999, TimeOffset=0, DeadTimeTable='MUSR62260_deadtime MA')
MuonGroupingAsymmetry(InputWorkspace='__MUSR62260_pre_processed_data', OutputWorkspace='MUSR62260; Group; top; Asymmetry; MA', OutputUnNormWorkspace='__MUSR62260; Group; top; Asymmetry; MA_unnorm', GroupName='top', Grouping='17-24,49-56', AsymmetryTimeMin=0.10199999999999999, AsymmetryTimeMax=32.293998718261719)
MuonGroupingAsymmetry(InputWorkspace='__MUSR62260_pre_processed_data', OutputWorkspace='MUSR62260; Group; bkwd; Asymmetry; MA', OutputUnNormWorkspace='__MUSR62260; Group; bkwd; Asymmetry; MA_unnorm', GroupName='bkwd', Grouping='25-32,41-48', AsymmetryTimeMin=0.10199999999999999, AsymmetryTimeMax=32.293998718261719)
MuonGroupingAsymmetry(InputWorkspace='__MUSR62260_pre_processed_data', OutputWorkspace='MUSR62260; Group; bottom; Asymmetry; MA', OutputUnNormWorkspace='__MUSR62260; Group; bottom; Asymmetry; MA_unnorm', GroupName='bottom', Grouping='1-8,33-40', AsymmetryTimeMin=0.10199999999999999, AsymmetryTimeMax=32.293998718261719)
MuonGroupingAsymmetry(InputWorkspace='__MUSR62260_pre_processed_data', OutputWorkspace='MUSR62260; Group; fwd; Asymmetry; MA', OutputUnNormWorkspace='__MUSR62260; Group; fwd; Asymmetry; MA_unnorm', GroupName='fwd', Grouping='9-16,57-64', AsymmetryTimeMin=0.10199999999999999, AsymmetryTimeMax=32.293998718261719)

# Dictionary { workspace_name: (workspace_index, start_x, end_x) }
input_data = {
    "MUSR62260; Group; bkwd; Asymmetry; MA": (0, 0.118000, 15.000000),
    "MUSR62260; Group; bottom; Asymmetry; MA": (0, 0.118000, 15.000000),
    "MUSR62260; Group; fwd; Asymmetry; MA": (0, 0.118000, 15.000000),
    "MUSR62260; Group; top; Asymmetry; MA": (0, 0.118000, 15.000000)
}

# Fit function as a string
function = \
    "name=GausOsc,A=0.2,Sigma=0.2,Frequency=1.4,Phi=0"

# Fitting options
max_iterations = 500
minimizer = "Levenberg-Marquardt"
cost_function = "Least squares"
evaluation_type = "CentrePoint"
output_base_name = "Output_Fit"

# Perform a sequential fit
output_workspaces, parameter_tables, normalised_matrices = [], [], []
for input_workspace, domain_data in input_data.items():
    fit_output = Fit(Function=function, InputWorkspace=input_workspace, WorkspaceIndex=domain_data[0],
                     StartX=domain_data[1], EndX=domain_data[2], MaxIterations=max_iterations,
                     Minimizer=minimizer, CostFunction=cost_function, EvaluationType=evaluation_type,
                     Output=output_base_name + input_workspace)

    output_workspaces.append(output_base_name + input_workspace + "_Workspace")
    parameter_tables.append(output_base_name + input_workspace + "_Parameters")
    normalised_matrices.append(output_base_name + input_workspace + "_NormalisedCovarianceMatrix")

    # Use the parameters in the previous function as the start parameters of the next fit
    function = fit_output.Function

# Group the output workspaces from the sequential fit
GroupWorkspaces(InputWorkspaces=output_workspaces, OutputWorkspace=output_base_name + "Workspaces")
GroupWorkspaces(InputWorkspaces=parameter_tables, OutputWorkspace=output_base_name + "Parameters")
GroupWorkspaces(InputWorkspaces=normalised_matrices, OutputWorkspace=output_base_name + "NormalisedCovarianceMatrices")

# Plot the results of the fit
fig, axes = plt.subplots(nrows=2,
                         ncols=len(output_workspaces),
                         sharex=True,
                         gridspec_kw={"height_ratios": [2, 1]},
                         subplot_kw={"projection": "mantid"})

for i, workspace_name in enumerate(output_workspaces):
    workspace = AnalysisDataService.retrieve(workspace_name)
    axes[0, i].errorbar(workspace, "rs", wkspIndex=0, label="Data", markersize=2)
    axes[0, i].errorbar(workspace, "b-", wkspIndex=1, label="Fit")
    axes[0, i].set_title(workspace_name)
    axes[0, i].set_xlabel("")
    axes[0, i].tick_params(axis="both", direction="in")
    axes[0, i].legend()

    axes[1, i].errorbar(workspace, "ko", wkspIndex=2, markersize=2)
    axes[1, i].set_ylabel("Difference")
    axes[1, i].tick_params(axis="both", direction="in")

fig.subplots_adjust(hspace=0)
fig.show()
```

robertapplin, 2022-06-14:
> Reference

This seems to only be an issue on IDAaaS. (which uses non-conda, so its probably not worth fixing this)

closed 2022-06-14 by robertapplin