mantidproject/mantid #34074: Ubuntu Interface Smoke Tests  :facepalm:

DanielMurphy22, 2022-06-14:
* Follow the instructions [here](https://download.mantidproject.org/ubuntu.html) to install the .deb for Ubuntu

**Before testing**:
- Check this testing issue relates to the OS you will test on.
- If unassigned, please assign yourself as for a normal Github issue.
- Please run these tests on the release package of Mantid; **not a locally built version**.

**Afterwards**:
- Comment below with any issues you came across.
- If no major issues were found, please **close this testing issue**.
- Check the main issue for this OS for other unassigned smoke tests.

If you have any questions please contact the creator of this issue.

 :facepalm:

## ISIS only, if possible, so catalogue access can be tested

- [ ] MantidWorkbench opens without errors or warnings 
- [ ] Every option in `Interface` opens a GUI 
- [ ] Load some test data 
- [ ] Access Catalogue through algorithms (use Facilities account, same as for IDAaaS): [CatalogLogin](https://docs.mantidproject.org/algorithms/CatalogLogin-v1.html) and [CatalogGetDataFiles](https://docs.mantidproject.org/algorithms/CatalogGetDataFiles-v1.html)
- [ ] Saving/loading projects works 
 - [ ] Alter preferences in [File > Settings](https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/useful/01_save_settings.html#settings) and check they are obeyed

gemmaguest, 2022-06-14:
The first time I tried to save the project, on clicking File->Save Project it gave this error:
```
 File "/home/gemma/Downloads/mantidworkbench/lib/python3.8/site-packages/workbench/app/mainwindow.py", line 666, in save_project
   self.project.save(CONF)
 File "/home/gemma/Downloads/mantidworkbench/lib/python3.8/site-packages/mantidqt/project/project.py", line 80, in save
   return self.open_project_save_dialog(conf)
 File "/home/gemma/Downloads/mantidworkbench/lib/python3.8/site-packages/mantidqt/project/project.py", line 103, in open_project_save_dialog
   ProjectSaveDialogPresenter(self, conf)
 File "/home/gemma/Downloads/mantidworkbench/lib/python3.8/site-packages/mantidqt/widgets/saveprojectdialog/presenter.py", line 36, in __init__
   self.view.set_save_altered_workspaces_only(project.save_altered_workspaces_only)
 File "/home/gemma/Downloads/mantidworkbench/lib/python3.8/site-packages/mantidqt/widgets/saveprojectdialog/view.py", line 33, in set_save_altered_workspaces_only
   self.save_altered_workspaces_only_radio_button.setChecked(check)
TypeError: setChecked(self, bool): argument 1 has unexpected type 'str'
```

I restarted mantid and after that it worked

robertapplin, 2022-06-14:
> The first time I tried to save the project, on clicking File->Save Project it gave this error:
>
> ```
>  File [...]

This can be fixed by changing these two lines to use the `type=bool` keyword:
https://github.com/mantidproject/mantid/blob/1ca967c21cfb5e9ce4a6f9f489b89c910842155b/qt/python/mantidqt/mantidqt/project/project.py#L58-L59

A similar fix is seen in https://github.com/mantidproject/mantid/pull/34020

robertapplin, 2022-06-14:
Closing this as all issues have been created: #34088