mantidproject/mantid #34077: Ubuntu Workspace/Algorithm Smoke Tests   :man_astronaut: :woman_astronaut: :space_invader:

DanielMurphy22, 2022-06-14:
* Follow the instructions [here](https://download.mantidproject.org/ubuntu.html) to install the .deb for Ubuntu

**Before testing**:
- Check this testing issue relates to the OS you will test on.
- If unassigned, please assign yourself as for a normal Github issue.
- Please run these tests on the release package of Mantid; **not a locally built version**.

**Afterwards**:
- Comment below with any issues you came across.
- If no major issues were found, please **close this testing issue**.
- Check the main issue for this OS for other unassigned smoke tests.

If you have any questions please contact the creator of this issue.

 :man_astronaut: :woman_astronaut: :space_invader:

* Load some workspaces, try 5 from different instruments
* Show detectors on these workspaces
* Check Instrument Viewer for these workspaces, follow  documentation for  [Instrument Viewer Widget](https://docs.mantidproject.org/workbench/instrumentviewer.html) and some more detail on the picking tools in the [Basic Course](https://docs.mantidproject.org/tutorials/mantid_basic_course/connecting_data_to_instruments/03_investigating_data.html)
  - [ ] Render tab 
  - [ ] Draw tab 
  - [ ] Pick tab 
  - [ ] Instrument tree tab 
* Test 5 - 10 random algorithms, follow [how to run Algorithms](http://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/algorithms_workspaces_and_histories/02_algorithms.html#algorithms) and [pick your favourite](https://docs.mantidproject.org/algorithms/)
 - [ ] Help button 
 - [ ] Validation of inputs 
 - [ ] Overwriting workspace 
 - [ ] Run the algorithm 
- [ ] Check right-click options of the workspaces, they are listed [in the documentation](https://www.mantidproject.org/MantidPlot:_The_Workspace_Menu#Save_Nexus).

darominski, 2022-06-14:
Instrument viewer issues:
- When a workspace is renamed, the title of the instrument viewer does not change. It does change for plots created from that workspace.
- Also, when the workspace is renamed and then deleted, the instrument viewer window stays open but is now blank. -> tested with `LOQ74044.nxs` from UnitTest directory.
- Hard crash observed when masking ILL instrument through the Draw tab, when the mask is applied to data. This was observed notably for IN5 and PANTHER, using data from `UnitTest/ILL` directory.  This was not observed for all ILL instruments if only a part of the instrument is masked, but all crash when all of pixels are marked. No issues when testing with empty instrument. Masking TOSCA loaded from TOSCA26911_show_detectors.nxs leads to a crash as well.

darominski, 2022-06-14:
Right click options of the algorithms:
- sample logs and workspace history windows persist after a workspace has been deleted - is this the expected behaviour?
- in the case of sample logs, the dialog title does not change when the workspace is renamed

DanielMurphy22, 2022-06-14:
> In the case described above, the scale has a tendency to change between Linear and SymmetricLog10 without updating the selection combobox.

How did you produce this??

darominski, 2022-06-14:
I admit this is a bit finicky to consistently reproduce, but we started observing the issue described above sporadically when looking at data in instrument viewer. Right now, I can reproduce swapping Linear with SymmetricLog10 each time the name is changed (but the combobox is properly updated in that case), and I am still trying to consistently reproduce the issue described. To have scales swapping:
1. load workspace, I used data from `LOQ74044.nxs`
2. show instrument
3. Change scale from the default selection to a different one (Linear->SymmetricLog10, and vice-versa)
4. Rename the workspace that is displayed, the scale should change back to the default choice.
5. Each subsequent rename should change the scale, and update the combobox.

darominski, 2022-06-14:
The most troubling issue is the masking. We can reproduce it locally using recent nightly version, and on a range of instruments.

darominski, 2022-06-15:
With the only critical issue found and addressed, I think I can safely close this issue now.