mantidproject/mantid #34079: Zoom on plot on MacOS shows box in wrong place

DanielMurphy22, 2022-06-14:
Found in Smoke testing: https://github.com/mantidproject/mantid/issues/34059#issuecomment-1154934444

**Describe the bug**
The zoom box is a little up and left of where I'm zooming.

**To Reproduce**
Download release candidate
open a plot of any data
click the zoom tool in the toolbar and drag a box
is it in the correct place?

**Screenshots**
[image]

**Platform/Version (please complete the following information):**
 - OS: MacOS
 - OS Version: Monteray
 - Mantid Version6.4 release candidate

**Additional context**
Only happens on a connected second screen!

DanielMurphy22, 2023-01-19:
Found again in manual testing for release 6.6

martyngigg, 2023-01-24:
I won't get time for this I'm afraid.

DanielMurphy22, 2023-01-26:
We should check if this is the case in standalone matplotlib outside of workbench

martyngigg, 2023-01-26:
Just for the record that while we discussing this we found that we are overriding the zoom rectangle behaviour in https://github.com/mantidproject/mantid/blob/main/qt/applications/workbench/workbench/plotting/mantidfigurecanvas.py#L19

closed 2023-02-01 by DannyHindson