mantidproject/mantid #34081: Instrument view window shrink

DanielMurphy22, 2022-06-14:
Related history: https://github.com/mantidproject/mantid/issues/29465 https://github.com/mantidproject/mantid/issues/28468

**Describe the bug**
On MacOS the instrument viewer displays in the bottom left corner of the display window. 
Regression from 6.3

**To Reproduce**
Have a second screen connected
Load any data
open the instrument viewer
select a display that is not Full3D (e.g. cylindrical)
Does it display in the bottom left or the full window??

See screenshots in linked issues at the top^^^

**Screenshots**

**Platform/Version (please complete the following information):**
 - OS: MacOS
 - OS Version: Monteray (all)
 - Mantid Version  6.4 candidate

DanielMurphy22, 2022-06-14:
First change I've noticed is the std::move here: https://github.com/mantidproject/mantid/pull/29432/files#diff-fd56035b44215d7b4bb9509a6a18bc5e519f0e7a1054f9655b9c97875b503531R50

DanielMurphy22, 2022-06-14:
I'm really now wondering is there an important change here: https://github.com/mantidproject/mantid/pull/29432/files#diff-372fb0b99cb65f96a2d4884969524209f0280a5d27732011712924f1cce09f20R1172-R1189

martyngigg, 2023-01-16:
This unfortunately still seems to be an issue on screens that report `devicePixelRatio: 2.0`. It's fine if the value is 1.0 here.

DanielMurphy22, 2023-01-17:
I also found this again and @gui-co mentioned a user at ILL has recently noticed this bug.
[image]

martyngigg, 2023-01-17:
I have found the issue here. I think I can put in a small fix for the release now that others are starting to see this.

closed 2023-02-08 by peterfpeterson